darthcloud/BlueRetro #567: On controller with report sent on change joystick calibration is delayed past connection

darthcloud, 2023-01-20:
### BlueRetro firmware version

all

### BlueRetro firmware specification

Select a spec

### BlueRetro firmware variant

Select a variant

### BlueRetro hardware type

Select a type

### Manufacturer

all

### System used

Select a system

### Bluetooth controller brand & name

xbox, stadia

### What is problem? (only list ONE problem per report)

On controller with report sent on change joystick calibration is delayed past connection

### What did you expect to happen?

We could either:

1. Figure a way to force a few report update at init
2. We could assume axes neutral is 0 past some X time past init

DJm00n, 2023-02-11:
>Figure a way to force a few report update at init
I think HID Get_Report(Input) request is a way to go.

darthcloud, 2023-04-19:
Get_Report did sound like the proper way to do it, event the HID spec suggest using this mechanism for initialization:

```
This request is useful at initialization time for absolute items and for
determining the state of feature items. This request is not intended to be used
for polling the device state on a regular basis.
```

Unfortunately the controllers for which I needed to do this do not implement it properly:

8bitdo will ignore the request:
https://github.com/darthcloud/bt_traces/blob/master/blueretro_8bitdo_sf30_get_report.pcapng

Retrobit answer with a weird report, all zeroes including report ID:
https://github.com/darthcloud/bt_traces/blob/master/blueretro_retrobit_saturn_get_report.pcapng

For BLE device I used the BT_ATT_OP_READ_REQ cmd directly on the report attribute.

For Xbox controller this result in a report that look right but in reality it's all hardcoded value, it doesn't reflect the real state of the various axis.
https://github.com/darthcloud/bt_traces/blob/master/blueretro_xbox_xs_ble_read_report.pcapng

For BLE device I think reading the report attribute directly is not the proper way of doing a "GET_REPORT"
I probably need to go through the HID Control Point attribute.

For Generic/Xbox BR/EDR devices I think I will simply remove the joystick center calib and assume middle to be exactly at the middle of the range.

For BLE I tempted to do the same but I might experiment with the HID Control Point first.

darthcloud, 2023-04-19:
Actualy base on https://www.bluetooth.com/specifications/specs/human-interface-device-service-1-0/ & https://www.bluetooth.com/specifications/specs/hid-over-gatt-profile-1-0/ HID Control Point is only for suspend cmds.

The various "legacy" HID cmds look to be absent from the LE HID specs.

So I will just remove joystick center calib for BLE device as well.

DJm00n, 2023-04-19:
>For BLE device I used the BT_ATT_OP_READ_REQ cmd directly on the report attribute.

it should work according to the HIDS spec...but seems these devices are not fully compliant.
[image]

darthcloud, 2023-04-19:
For now I will maintain current behavior and simply add a note in the wiki

https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#ii---blueretro-joystick-calibration

I might hardcode the neutral value if I get too much complain in the future

closed 2023-04-19 by darthcloud