darthcloud/BlueRetro #576: BlueRetro PS2, ResidenEvil4 USA Rumble Error.

mmejiaherrera, 2023-01-22:
### BlueRetro firmware version

v1.8.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

DualShock4

### What is problem? (only list ONE problem per report)

When activating the Rumble function and playing ResidentEvil4 USA on PS2, with the DualShock4 controller, the vibration activates uncontrollably during the menu and all cinematics. Also disabling the vibration in the game options does not work.

### What did you expect to happen?

That the vibration worked correctly in the game.

ButtonBasher92, 2024-01-06:
I too have this problem in Resident Evil 4 PS2 using a DS3 & DS4 controller.

darthcloud, 2024-05-29:
Can youtest this beta, this add dual motor and variable force support for Playstation and Dreamcast
I think it would fix that issue aswell

https://github.com/darthcloud/BlueRetro/actions/runs/9286878322

darthcloud, 2024-10-06:
Fixed in v24.10:
https://github.com/darthcloud/BlueRetro/releases/tag/v24.10

closed 2024-10-06 by darthcloud