darthcloud/BlueRetro #577: Ridge Racer V, PS2, 8BitDo Pro2

L10N37, 2023-01-24:
### BlueRetro firmware version

1.8.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Chinese, 2 different builds , same problem

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

8BitDo Pro2

### What is problem? (only list ONE problem per report)

When you get into a race, the accelerator (X button) is pressed and held by the BlueRetro firmware, not physically.
The car redlines, then for some reason, X button stops working altogether. 
The game isn't playable.
I chose manual and confirmed you can change gears up and down.

### What did you expect to happen?

That the buttons keep working once a race starts

darthcloud, 2023-10-07:
I tried this game with OPL and it worked fine for me.

Can you retry with latest v1.9 beta?
https://github.com/darthcloud/BlueRetro/actions/runs/6437347093

darthcloud, 2024-04-20:
Feel free to reopen if you retest and still have the issue.

closed 2024-04-20 by darthcloud