darthcloud/BlueRetro #579: Gamecube 3rd Party controller issues.

Alex13mod, 2023-01-24:
### BlueRetro firmware version

1.8.2

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install modkit

### Manufacturer

LaserBear

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Keyboard 

### What is problem? (only list ONE problem per report)

Plugging the Retro Fighters in Port1 and using Bluetooth Keyboard. 
Keyboard stays in P1 and doesnt switch to P2 causing conflict with 3rd party controller.

### What did you expect to happen?

Keyboard synch to p2 and 3rd party wireless con troller connecte to P1 
2.4ghz controller
*Blade GC*

Alex13mod, 2023-01-24:
Heres a video

Sauced, 2023-01-25:
This is due to the Blade controller not triggering controller detection. 
A modification of the Retro Fighters receiver is required to trigger port detection. 
The shield on the controller connector needs to be connected to the ground connection on the receiver board.

Alex13mod, 2023-01-25:
> This is due to the Blade controller not triggering controller detection. A modification of the Retro Fighters receiver is required to trigger port detection. The shield on the controller connector needs to be connected to the ground connection on the receiver board.

Thanks  for the update!!

z3razerviper, 2023-01-30:
Just got my battler gc and it does not work correctly at all the analogue sticks keep getting stuck. You cant even make it off the first screen in zelda four swords.

z3razerviper, 2023-01-30:
> This is due to the Blade controller not triggering controller detection. A modification of the Retro Fighters receiver is required to trigger port detection. The shield on the controller connector needs to be connected to the ground connection on the receiver board.

Do you have any links describing how to do the mod?

Alex13mod, 2023-01-30:
By Laserbear
[image]
[image]

z3razerviper, 2023-01-30:
Do you know if this will solve the wierd issues with sticks getting all confused.

Sauced, 2023-01-30:
It should, you can test for sure by putting blue Retro into sleep at boot by holding reset. port 4 will be solid on and the blue retro board will not communicate on any ports. The communication is disabled when a controller is detected so it may be a control conflict.

Alex13mod, 2023-01-30:
> Do you know if this will solve the wierd issues with sticks getting all confused.

Yes, The issues is port detection cause you probably using a a dongle and a Bluetooth device and both are conflicting with Port1

Alex13mod, 2023-01-30:
> Do you know if this will solve the wierd issues with sticks getting all confused.

Here's a video of me using the mod:
[BLADE GC](https://twitter.com/alex13mod/status/1618349354963832832?t=B9S22B2tOBjwD66i9k77IQ&s=19)

z3razerviper, 2023-01-30:
I will try it later however I am not using it with a Bluetooth device that's why I am a little worried. :(

z3razerviper, 2023-01-31:
Ok so I tried the fix and I also tried disabling bluetooth on the blueretro no change I think i just have a bad controller :|

Alex13mod, 2023-01-31:
> Ok so I tried the fix and I also tried disabling Bluetooth on the blueretro no change I think i just have a bad controller :

> Ok so I tried the fix and I also tried disabling Bluetooth on the blueretro no change I think i just have a bad controller :|

When you plug the dongle, does the Blue light switch to second port? Have you tried it on a diffrent system or PC?

z3razerviper, 2023-01-31:
Yes it does light blue. I tried it on a pc but did not have any gamecube games to test it with since I have never setup dolphin. My primary use of this controller was only for gamecube.

Sauced, 2023-01-31:
If it was still doing it with the blue retro disabled I would lean towards a bad controller/dongle over anything else. If a wired controller has the same issue then maybe it would be the board.

closed 2023-04-01 by darthcloud