darthcloud/BlueRetro #584: [Gamecube internal] Xbox One controllers seem to have compatabillity issues

T1GWBJAH, 2023-01-27:
### BlueRetro firmware version

1.8.2

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install modkit

### Manufacturer

Laserbear

### System used

Nintendo GameCube

### Bluetooth controller brand & name

wireless controller for xbox one model 1708

### What is problem? (only list ONE problem per report)

Connected controller to gamecube, and X button would not act as the B button like it usually does. 

### What did you expect to happen?

For the X Button to act as B button on original Gamecube controller

quentinmaisonneuve, 2023-02-02:
I had the same issue, I solved it by setting the mapping config to GameCube Merge analog &amp; digital trigger on the setting page : https://blueretro.io/presets.html
[image]

qwertymodo, 2023-02-19:
There are several button mappings on the XBone controller that don't seem to match the documented ones [here](https://docs.google.com/spreadsheets/d/e/2PACX-1vT9rPK2__komCjELFpf0UYz0cMWwvhAXgAU7C9nnwtgEaivjsh0q0xeCEiZAMA-paMrneePV7IqdX48/pubhtml)

Here are the mapping issues I see:

XBone Button | Mapped GC Button | Expected GC Button
X   |   Y   |   B
Y   |   Z   |   Y
RB   |   Start   |   Z
Menu   |   None   |   Start

The rest of the buttons seem to be correctly mapped (A->A, B->X, L/R triggers, D-pad)

darthcloud, 2023-02-19:
Is your Xbox controller having latest FW installed?

qwertymodo, 2023-02-19:
It was not.  The firmware update seems to have fixed my issue (though, I also applied the GameCube Merge analog & digital trigger profile shown above, so that might also be necessary).  I only ever use this XBone controller on the PC, so I wasn't aware that firmware updates were a thing, and had to download an app to apply the update.

darthcloud, 2024-04-20:
@T1GWBJAH you still have that issue? Try updating the Xbox controller firmware via the [Xbox accessories Win10 app](https://apps.microsoft.com/store/detail/xbox-accessories/9NBLGGH30XJ3) as suggested.

darthcloud, 2024-10-06:
Feel free to reopen if you got more info

closed 2024-10-06 by darthcloud