darthcloud/BlueRetro #586: PowerA Model# 1511638-01 .pcapng trace

bigalsrover, 2023-01-28:
### BlueRetro firmware version

1.8.2

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install modkit

### Manufacturer

LaserBear

### System used

Nintendo GameCube

### Bluetooth controller brand & name

PowerA Model 1511638-01

### What is problem? (only list ONE problem per report)

Controller would become unresponsive as a player 2 input with a Switch Po Controller connected as Player 1 

### What did you expect to happen?

Both controllers connected at the same time

Order of buttons pressed during trace:

1 -
2 +
3 Capture
4 Home
5 Y
6 X
7 A
8 B
9 C-stick
10 D-Pad (UP)
11 D-Pad (DOWN)
12 D-Pad (LEFT)
13 D-Pad (RIGHT)
14 D-Pad (NE)
15 D-Pad (SW)
16 D-Pad (NW)
17 D-Pad (SE)
18 Left Stick
19 Left Shoulder
20 Right Shoulder
21 ZR
22 ZL
23 Pair button

### Attach files like logs or Bluetooth traces here

[BT_Trace2_PowerA-Model_1511638-01.zip](https://github.com/darthcloud/BlueRetro/files/10525997/BT_Trace2_PowerA-Model_1511638-01.zip)

TriVoxel, 2023-03-25:
I think I am experiencing a similar issue. Hopefully this gets looked into as these are the perfect controllers... :(

darthcloud, 2023-03-25:
Duplicate of #146

closed 2023-03-25 by darthcloud