darthcloud/BlueRetro #588: IFYOO-WAX1 Gbros type adapter with Wii Classic Controller, pairing issue

klunkee, 2023-01-29:
### BlueRetro firmware version

1.8.2

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Self

### System used

Nintendo GameCube

### Bluetooth controller brand & name

IFYOO-WAX1 Gbros type adapter with Wii Classic Controller attached

### What is problem? (only list ONE problem per report)

With X-input mode on the IFYOO, adapter connects to BlueRetro and all buttons work for 30 seconds but cannot maintain pairing. The light on the Gbros clone keeps blinking like it was not fully paired and then disconnects. 

Using Switch mode, adapter pairs and remains paired but R and L triggers are mapped to Z button and ZR and ZL are triggers. Both sticks do not work but dpad and all other buttons seem to work.

### What did you expect to happen?

If the adapter can pair correctly in X-input mode, then all inputs would be working as intended. If Switch mode inputs are remapped and stick input recognized it would also work.

Order of buttons pressed:

Wii Classic Controller

1. L
2. ZL
3. ZR
4. R
5. Dpad Up
6. Dpas Right
7. Dpad Down
8. Dpad Left
9. Leftstick Up
10. Leftstick Right
11. Leftstick Down
12. Leftstick Left
13. Minus(Select)
14. Home
15. Plus(Start)
16. Rightstick Up
17. Rightstick Right
18. Rightstick Down
19. Rightstick Left
20. A
21. B
22. X
23. Y

### Attach files like logs or Bluetooth traces here

[IFYOO-WAX1 Xinput Wii Classic Controller.zip](https://github.com/darthcloud/BlueRetro/files/10528998/IFYOO-WAX1.Xinput.Wii.Classic.Controller.zip)

darthcloud, 2023-10-08:
The Xinput mapping look to be fine.

Unfortunately the PC trace can't help much debug disconnection issue on BlueRetro.

Try latest beta to see if you still got issue first:
https://github.com/darthcloud/BlueRetro/actions/runs/6443343629

If your BlueRetro got a USB port, try to reproduce problem while you are saving the USB logs:
https://github.com/darthcloud/BlueRetro/wiki/Getting-BlueRetro-debug-logs-via-Serial-port-Windows-10

darthcloud, 2024-10-06:
Feel free to reopen if you got more info

closed 2024-10-06 by darthcloud