darthcloud/BlueRetro #600: [N64] Conflict With Original N64 Controllers – Nintendo Switch Online

danny-ml, 2023-02-06:
### BlueRetro firmware version

1.8.2

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

Generic ESP32 DEVKIT V1

### System used

Nintendo 64

### Bluetooth controller brand & name

[Original] N64 - Nintendo Switch Online Controller Mod. HAC-043

### What is problem? (only list ONE problem per report)

Random disconnections. Conflict when connecting 3 or 4 equal controls. 

### What did you expect to happen?

With 4 synchronized controllers brand 8Bitdo Model SN30 Pro in Switch mode, I don't have that problem. I think it is a bug in the firmware.

### Attach files like logs or Bluetooth traces here

[N64 Ctrlr BT.zip](https://github.com/darthcloud/BlueRetro/files/10667624/N64.Ctrlr.BT.zip)

JamesFromCC, 2023-04-13:
I've been having this same issue! I've been trying for months to come up with a solution that ideally isn't just using 4 blueretros. 

2 NSO controllers seems ok but when you try to connect a 3rd it boots one off and you can only maintain a max of 2 connected. It also gets kind of buggy when you try to do this. I believe after such an attempt it showed controller 1 and 3 as connected via the indicator lights and I noticed 1 of the 2 controllers connected had BAD latency to varying degrees. So realistically I don't trust it to work with more than 1 reliably, currently. I'm willing to help beta test firmware as I have 4 of these controllers. I'd also be willing to make a monetary contribution to whomever maintains the code.

As someone who has used the NSO controllers a bit, here are some notes I have. I've noticed MAJOR issues when trying to use the NSO controller on android as well, and the connection issues may be related. I say this because the laggy behaver where it seems to update input only every 500ms. So there seems to be something going on with the NSO controllers firmware not playing well with other devices. But I can note 4 controllers seemed to work "ok" on in windows. I noticed minor latency on one controller when all 4 were used to play games, but me and my 3 brothers were able to enjoy a few rounds on smash in emulation; If the blue retro worked at least that well, it wouldn't be that bad.

IDK exactly what's going on with the blueretro but I also think it's solvable in firmware. Also to note, I have tried 3 different blueretro adapters (including an official) and multiple firmware versions thinking I did something wrong early on. 

I wish it worked like it's supposed to, because it seems pretty awesome when it's working for other stuff. I appreciate the level of effort that has already been put in to support this and hope to see it's potential realized very soon. I desperately want to add a blueretro internal to my N64, but until this is working, I'm torn on what to do.

*NSO = Nintendo Switch Online 64

Traxus7, 2023-05-12:
Definitely issue #146. I'm hoping for a fix one day.

darthcloud, 2024-04-20:
For one player random disconnect that should have been fixed last fall, try v1.9.2.

Otherwise multiplayer issue are related to #146

closed 2024-04-20 by darthcloud