darthcloud/BlueRetro #603: [GC] HORI Battle pad no rumble

milouzzzz, 2023-02-09:
### BlueRetro firmware version

1.8.1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

SP-cow ESP32S Dev board

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Switch HORI battle pad 

### What is problem? (only list ONE problem per report)

Rumble doesn't seems to work.
Rumble is activated in the web interface.

Tried as well with other controllers (8bitdo sn30 pro, wiiu pro controller and a switch pro) but i never got rumble to work. I think i missed something somewhere.

I've installed it on 2 differrent gamecubes but none of them rumble.

### What did you expect to happen?

That rumble works :)

sapkyy, 2023-04-09:
Hello! But Hori battle pad doesn't have any rumble function..

darthcloud, 2025-01-24:
Not sure about the spec of that controller if it got rumble or not.
IF so likely [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01) would work better.

closed 2025-01-24 by darthcloud