darthcloud/BlueRetro #604: Dont have rumble in new version PS1/2 gamepad xbox one

MaxJumper, 2023-02-11:
### BlueRetro firmware version

1.8.2

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

i am

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

XBOX one 1708

### What is problem? (only list ONE problem per report)

Dont have rumble in new version

### What did you expect to happen?

I test version 1.5 - only one motor work, but work fine (full power)
1.7.3 and 1.8.2 dont have rumble.
I'm testing with a gamepad tester on PS2. And there is an opportunity to turn on different vibration motors at different speeds. So on the new versions, when I run the vibration test, the gamepad emits a barely noticeable sound that does not change in any way and does not stop.

MaxJumper, 2023-02-11:
I checked on the wii remote - the situation repeated itself: vibration works on the old firmware 1.5, but there is no vibration on the new ones

DJm00n, 2023-02-16:
@MaxJumper have you tried to update your controller's firmware?
https://support.xbox.com/en-US/help/hardware-network/controller/update-xbox-wireless-controller

MaxJumper, 2023-02-16:
> @MaxJumper have you tried to update your controller's firmware? 
@DJm00n Are you serious? Do you recommend updating the remote from the Nintendo Wii too?
The problem is that at 1.5 vibration works on both gamepads. And it does not work on the new B.R. firmware.

darthcloud, 2023-04-01:
Are you rumble is enable in config?
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#26---example-enable-rumble

It works fine for me

MaxJumper, 2023-04-05:
> Are you rumble is enable in config?
This is joke ?
Of course I turned on vibration in the settings. And I tried different firmware, and I indicated this in the description!

darthcloud, 2023-04-05:
Bad attitude so closed, probably duplicate of previous rumble anyway

closed 2023-04-06 by darthcloud