darthcloud/BlueRetro #607: Ape Escape doesn’t work with BlueRetro PS1/2 Adapter

BoxIsMyNamee, 2023-02-12:
### BlueRetro firmware version

1.72

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Ps3, Ps5.

### What is problem? (only list ONE problem per report)

Ape Escape on PS1 requires a Dual Shock 1 to play it. It’s works with no other controller. And this adapter doesn’t account for it. Is there a firmware update I could do to get this fixed? One of my favs is ape escape. 

### What did you expect to happen?

It to work normally. Or at least account and use a Dual Shock 1 by default considering it’s the same as a dualshock 2. and would work with ps2 games.

darthcloud, 2023-02-12:
BlueRetro emulate a DualShock 2 and I personally test Ape Escape every release on a PS1.

If you are using a PS2 it's probably aproblem related to the emulator.

BoxIsMyNamee, 2023-02-15:
> BlueRetro emulate a DualShock 2 and I personally test Ape Escape every release on a PS1.
> 
> If you are using a PS2 it's probably aproblem related to the emulator.

Can I change what controller the blueretro adapter identifies as? Like instead of emulating dualshock 2, emulate dualshock 1??

BoxIsMyNamee, 2023-02-18:
my bad, my disc was scratched. getting it resurfaced made the game launch fine. i’m a dunce lmfaoo

closed 2023-02-18 by BoxIsMyNamee