darthcloud/BlueRetro #617: GuliKit Controller XW (KingKong 2) pair trace

rob-farrellrobotics, 2023-02-18:
### BlueRetro firmware version

NA

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

me

### System used

Handheld

### Bluetooth controller brand & name

GuliKit Controller XW (KingKong 2)

### What is problem? (only list ONE problem per report)

New controller

### What did you expect to happen?

nothing, new controller

### Attach files like logs or Bluetooth traces here

[GuliKit Controller XW King Kong 2 pair trace.zip](https://github.com/darthcloud/BlueRetro/files/10774522/GuliKit.Controller.XW.King.Kong.2.pair.trace.zip)

anherooftime69, 2023-02-21:
#146 ?

darthcloud, 2023-10-08:
I don't really see anything wrong in the trace, look like standard Xinput.
What was the issue exactly?

You might want to try latest beta:
https://github.com/darthcloud/BlueRetro/actions/runs/6443343629

darthcloud, 2024-04-20:
Feel free to reopen when you can provide requested info.

closed 2024-04-20 by darthcloud