darthcloud/BlueRetro #622: Automatic connection does not occur with Stadia Controller

DJm00n, 2023-02-26:
### BlueRetro firmware version

1.8.3

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

BlueRetro Gaming Store

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Google Stadia Controller

### What is problem? (only list ONE problem per report)

I have successfully paired Stadia Controller to the BlueRetro (by holding Stadia button + the Y button) but controller does not connect automatically when turned on again and I have to pair them again each time.

Also checked the automatic bluetooth connection to the Windows PC (with "Realtek Bluetooth 5.0 Adapter" [0BDA:8771]) - and everything works fine there.

What other information can I provide?

### What did you expect to happen?

Paired controller should automatically connect to BlueRetro when switched on.

### Attach files like logs or Bluetooth traces here

Here is Wireshark dump from the PC<=>Stadia Controller - this is the moment of automatic connection of already paired controller:  [stadia_bluetoothle.pcapng.gz](https://github.com/darthcloud/BlueRetro/files/10834468/stadia_bluetoothle.pcapng.gz)

DJm00n, 2023-02-26:
Related: https://www.reddit.com/r/Stadia/comments/117pjwl/stadia_controller_bluetooth_mode_pairs_and_works/

haydnkilyan, 2024-09-08:
Stadia controllers do this for me too - GC internal mod

darthcloud, 2025-01-24:
Fixed in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)

closed 2025-01-24 by darthcloud