darthcloud/BlueRetro #638: GamePadALT doesn't seem to work for Sega Genesis

vicsidious, 2023-03-11:
### BlueRetro firmware version

1.8.3

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

Myself

### System used

SEGA MegaDrive Genesis

### Bluetooth controller brand & name

8BitDo Arcade Stick / 8BitDo DIY Kit for Sega Genesis

### What is problem? (only list ONE problem per report)

Selecting GamePadAlt and power cycling the Esp32 doesn't change to 6 buttons gamepad for Sega Genesis.
Firmware 1.7.3 didn't have this issue. I noticed it since I've updated to 1.8.3. I went from 173 to 183 directly.

### What did you expect to happen?

Enabling X, Y, Z and Mode button.

### Attach files like logs or Bluetooth traces here

I don't know how to do that, but am willing to learn!

retrofank, 2023-04-10:
1.8.3 works for Sony official DS4 controllers in ALT mode 6 button and a 8bitDo M30 i tried.  Make sure the setting in the web GUI is saved and a hard power cycle (not via button hold) is done.

Chandler-Kluser, 2023-05-28:
I am using v1.8.3 HW1 with 8BitDo M30. GamePad and GamePadAlt have only Start button working, I am using the BlueRetro Core from AliExpress with the MD adapter. Using DS4 is working as expected.

darthcloud, 2023-10-15:
Please try in v1.9
https://github.com/darthcloud/BlueRetro/releases/tag/v1.9

Tested in MK1 two player gamepadalt myself today.

Raton-Laveur, 2023-10-20:
Different behavior on 1.9.  X, Y and Z are ok, but Start and Mode are weird: 

BlueRetro firmware version
1.9

BlueRetro firmware specification
HW1

BlueRetro firmware variant
Universal

BlueRetro hardware type
External adapter with detachable cord

Manufacturer
Aliexpress

System used
SEGA Mega Drive II

Bluetooth controller brand & name
2x 8BitDo M30

What is problem? (only list ONE problem per report)
On both Player 1 and 2: 
Pressing Start gives Start + Mode 
Pressing Select gives nothing. 

What did you expect to happen?
Pressing Start giving Start
Pressing Select giving Mode.

scrap-a, 2024-04-19:
I also experienced the symptom of mixed START and MODE. (v1.9.2)
I am just guessing from looking at the source, as I think the cause is that START and MODE are in the same place in the mapping table of "genesis_btns_mask" in "main/adapter/wired/genesis.c".

        /* TH LOW */
        {
            0, 0, 0, 0,
            0, 0, 0, 0,
            0, 0, BIT(P1_LD_DOWN), BIT(P1_LD_UP),
            0, 0, 0, 0,
            BIT(P1_A), 0, 0, 0,
            BIT(P1_START), 0, 0, 0,
            0, 0, 0, 0,
            0, 0, 0, 0,
        },
        /* 6BTNS */
        {
            0, 0, 0, 0,
            0, 0, 0, 0,
            0, 0, 0, 0,
            0, 0, 0, 0,
            0, 0, 0, BIT(P1_Y),
            BIT(P1_MODE), 0, 0, 0,
            BIT(P1_X), BIT(P1_X), 0, 0,
            BIT(P1_Z), BIT(P1_Z), 0, 0,
        },
The BIT(P1_START) for TH LOW and the BIT(P1_MODE) for 6BTNS are both located in row 6, column 1.
I believe that the BIT(P1_MODE) for 6BTNS is probably correct in row 2.
`            0, BIT(P1_MODE), 0, 0,`

darthcloud, 2024-04-19:
Very good catch @scrap-a 

Feel free to send a PR or I can fix it later today.

scrap-a, 2024-04-19:
Unfortunately, I have not the build environment, so I cannot verify this right away.
If you need a PR, please wait a few days.

darthcloud, 2024-04-20:
That beta should fix the overlap between start and mode:
https://github.com/darthcloud/BlueRetro/actions/runs/8760755824

scrap-a, 2024-04-20:
Excellent!
The OVERLAP symptom has been completely fixed about both 1P and 2P.

closed 2024-04-24 by darthcloud