darthcloud/BlueRetro #645: Files for APEX 3 - Controller Mapped

hardlevel, 2023-03-17:
### BlueRetro firmware version

latest

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

gamescare

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

flydigi apex 3

### What is problem? (only list ONE problem per report)

No problem, I just mapped this amazing controller to be included on a future firmware

### What did you expect to happen?

blueretro supporting apex 3

### Attach files like logs or Bluetooth traces here

[apex3.txt](https://github.com/darthcloud/BlueRetro/files/10996753/apex3.txt)
[apex3.zip](https://github.com/darthcloud/BlueRetro/files/10996755/apex3.zip)

hardlevel, 2023-04-05:
hi friend, how are you? any chances of adding this controller? I sent the required files but until now, any answer

hardlevel, 2023-04-15:
would be nice at least an answer...

darthcloud, 2023-04-16:
The way request work is that you wait patiently for me to get to it.

I got limited free time and more important things to do.

darthcloud, 2025-01-24:
I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

Thanks

hardlevel, 2025-01-24:
sure friend, I'm glad that you can take a look at this.

I have a lot of new controllers that I can provide the info that you need

I have now from flydigi:
Direwolf 1
Direwolf 2
Direwolf 3
Vader 3
Apex 3
Apex 4

And from 8bitdo the new ultimate c

i will try to collect the data this weekend

hardlevel, 2025-01-24:
here friend, please take a look if the dump is good

[br_debug_trace.zip](https://github.com/user-attachments/files/18542337/br_debug_trace.zip)

darthcloud, 2025-01-25:
In what order did you press the buttons?

hardlevel, 2025-01-25:
i made it again to be sure of the order

this is the order:
d-pad-up
d-pad-down
d-pad-left
d-pad-right
l-up
l-down
l-left
l-right
r-up
r-down
r-left
r-right
L1
L2
L3
R1
R2
R3
X
Y
A
B
select
start
home

[br_debug_trace.zip](https://github.com/user-attachments/files/18547610/br_debug_trace.zip)

darthcloud, 2025-02-01:
Can you try this beta:
https://github.com/darthcloud/BlueRetro/actions/runs/13083210355

make sure you disable debug mode before updating

closed 2025-02-17 by darthcloud

hardlevel, 2025-02-18:
hi friend, really sorry, I didn´t saw your update with the beta update, I will try it as soon as possible, sorry again for the late reply