darthcloud/BlueRetro #646: Axis -> Button threshold doesn't work when set to 100

coolbho3k, 2023-03-17:
### BlueRetro firmware version

Current master

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Expressif

### System used

Nintendo GameCube

### Bluetooth controller brand & name

DualSense

### What is problem? (only list ONE problem per report)

I reproduced this with a DualSense, but I'm not sure if it happens in all situations.

When an axis is mapped to a button, the threshold is not possible to reach if the threshold is equal to 100.

I think I might see the bug, but I"m not sure. The logic for detecting the threshold is as follows:
```
int32_t threshold = (int32_t)(((float)map_cfg->perc_threshold/100) * ctrl_input->axes[src_axis_idx].meta->abs_max);
/* Check if axis over threshold */
if (abs_src_value > threshold) {
    out->btns[dst_btn_idx].value |= dst_mask;
    out->btns[dst_btn_idx].cnt_mask[dst & 0x1F] = map_cfg->turbo;
}
```

Let's say `perc_threshold` is 100, and `abs_max` is 0xFF.

`(int)(((float)100/100) * 0xFF)` is 0xFF. However, the maximum value that `abs_src_value` can be is 0xFF. Thus, `abs_src_value` will never be `> threshold`.

I think this can be fixed if we change the check to `if (abs_src_value >= threshold)`. It'll make the button slightly more sensitive.

### What did you expect to happen?

When the threshold is set to 100, the button should count as pressed when the axis position is at abs_max.

darthcloud, 2023-03-17:
If you want you can make a PR for ```if (abs_src_value >= threshold)``` and I will merge ASAP

Or I can do it later.

coolbho3k, 2023-03-17:
Sure, will make it tonight. Thanks!

PR #648 (2023-03-17): Fix axis to button mapping when threshold is 100%
Fixes #646.
closed 2023-03-17 by darthcloud