darthcloud/BlueRetro #652: Archivo control Kanji ps4

rorosala, 2023-03-20:
### BlueRetro firmware version

1.8.3

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

yo mismo

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Kanji

### What is problem? (only list ONE problem per report)

Quiero que este controlador sea compatible con bluretro  

### What did you expect to happen?

poder emparejar este controlador  

### Attach files like logs or Bluetooth traces here

[control kanji ps4.pcapng.gz](https://github.com/darthcloud/BlueRetro/files/11022804/control.kanji.ps4.pcapng.gz)
[texto.txt](https://github.com/darthcloud/BlueRetro/files/11022806/texto.txt)

yakaracolombia, 2023-03-23:
así no funcionan los "issues" en GitHub, si quieres implementar tu código y ya tienes tu version compatible debes crear un full request desde tu fork

darthcloud, 2025-04-16:
Feel free reopen once you got the trace

closed 2025-04-16 by darthcloud