darthcloud/BlueRetro #658: Wireless controller T-28 not syncing

tech4bot, 2023-03-25:
### BlueRetro firmware version

1.8.3

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

BlueRetro DIY ESP

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Wireless controller T-28

### What is problem? (only list ONE problem per report)

I've tried to reset and hold home and share button to put it on pair mode, the light flashes but never sync with blueretro. The controller is a clone for PS4, it works fine on PS4 and PC over bluetooth when I pair.

### What did you expect to happen?

Sync with blueretro.

### Attach files like logs or Bluetooth traces here

[wireless_controller.pcapng.zip](https://github.com/darthcloud/BlueRetro/files/11068348/wireless_controller.pcapng.zip)

tech4bot, 2023-03-25:
I looked over the serial of blueretro and I am getting the following:

# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_INQUIRY_RESULT
# Number of responce: 1
# bt_hci_cmd_connect
# Inquiry dev: 0 type: 0 bdaddr: 98:B6:E9:BD:23:25
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_INQUIRY_RESULT
# Number of responce: 1
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_CONN_COMPLETE
# dev: 0 acl_handle: 0x0081
# bt_hci_cmd_le_set_adv_disable
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# bt_type_update: dev: 0 type: 4 subtype: 0
# bt_hci_cmd_exit_periodic_inquiry
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_passive
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_auth_requested
# dev: 0 type: 4:0 Wireless Controller
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_LINK_KEY_REQ
# bt_hci_cmd_link_key_neg_reply
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_IO_CAPA_REQ
# bt_hci_cmd_io_capability_reply
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_USER_CONFIRM_REQ
# bt_hci_cmd_user_confirm_reply
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_LINK_KEY_NOTIFY
# BT_HCI_EVT_AUTH_COMPLETE
# dev: 0 Pairing done
# bt_l2cap_cmd_hid_ctrl_conn_req
# BT_HCI_EVT_DISCONN_COMPLETE
# DISCONN from dev: 0
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_active
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_periodic_inquiry
# bt_hci_cmd_le_set_adv_enable
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT

tech4bot, 2023-03-25:
Over serial I am getting:
```
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_INQUIRY_RESULT
# Number of responce: 1
# bt_hci_cmd_connect
# Inquiry dev: 0 type: 0 bdaddr: 98:B6:E9:BD:23:25
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_INQUIRY_RESULT
# Number of responce: 1
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_CONN_COMPLETE
# dev: 0 acl_handle: 0x0081
# bt_hci_cmd_le_set_adv_disable
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# bt_type_update: dev: 0 type: 4 subtype: 0
# bt_hci_cmd_exit_periodic_inquiry
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_passive
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_auth_requested
# dev: 0 type: 4:0 Wireless Controller
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_LINK_KEY_REQ
# bt_hci_cmd_link_key_neg_reply
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_IO_CAPA_REQ
# bt_hci_cmd_io_capability_reply
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_USER_CONFIRM_REQ
# bt_hci_cmd_user_confirm_reply
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_LINK_KEY_NOTIFY
# BT_HCI_EVT_AUTH_COMPLETE
# dev: 0 Pairing done
# bt_l2cap_cmd_hid_ctrl_conn_req
# BT_HCI_EVT_DISCONN_COMPLETE
# DISCONN from dev: 0
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_active
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_periodic_inquiry
# bt_hci_cmd_le_set_adv_enable
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
```

darthcloud, 2025-01-24:
I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

Thanks

tech4bot, 2025-02-02:
sure. I will install it and use the debug.

darthcloud, 2025-04-16:
Feel free reopen once you got the trace

closed 2025-04-16 by darthcloud