darthcloud/BlueRetro #659: PS3 controller not working

tech4bot, 2023-03-25:
### BlueRetro firmware version

1.8.3

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

BlueRetro DIY ESP

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Original Sony DualShock 3 SIXAXIS

### What is problem? (only list ONE problem per report)

After changing the mac on PC, it gets paired with blueretro, but no player light is selected.

### What did you expect to happen?

After successful pair player 1 led should light up. 

### Attach files like logs or Bluetooth traces here

```
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_CONN_REQUEST
# bt_hci_cmd_accept_conn_req
# Page dev: 0 type: 0 bdaddr: 41:42:5F:42:BD:E8
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_LE_ADVERTISING_REPORT
# BT_HCI_EVT_ROLE_CHANGE
# BT_HCI_EVT_CONN_COMPLETE
# dev: 0 acl_handle: 0x0080
# bt_hci_cmd_le_set_adv_disable
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_STATUS
# BT_L2CAP_CONN_REQ
# BT_L2CAP_CONF_REQ
# BT_L2CAP_CONF_RSP
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# bt_type_update: dev: 0 type: 4 subtype: 0
# bt_hci_cmd_exit_periodic_inquiry
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_passive
# bt_hci_cmd_le_set_scan_enable
# dev: 0 type: 4:0 Wireless Controller
# BT_L2CAP_CONN_REQ
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_L2CAP_CONF_REQ
# BT_L2CAP_CONF_RSP
# bt_hid_ps_init
# bt_hid_ps5_init_callback
```

darthcloud, 2023-03-25:
Base on log that doesn't look like an original PS3 controller, look like a PS4 one

darthcloud, 2023-03-25:
You did pair the MAC address with sixaxis tool first?

https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#211---windows

tech4bot, 2023-03-25:
Yes, I've configured it with the sixaxis tool and set the mac access of retropie. on the label of the controler is saying Sony model no CECHZC2E A1 if is relevant.

darthcloud, 2023-03-25:
ok but your log show no sign of a PS3 controller. Do your pc tool paring again and make sure you use BlueRetro mac address found on the web config page.

tech4bot, 2023-03-25:
I did a reset of the PS3 controller and configure the mac address again on sixaxis and now, after reset, I still cannot see player 1 led lighting up, but is just scrolling down thru menu on PS2 without pressing any buttons. If I use the left analog stick, it slows down the scrolling thru menu. This controller works fine on PS3 console, and buttons are ok and the mapping is ok, especially that PS button works as expected, as on PS3 console if you use any other controller then PS3, the home button will not work. This DIY blueretro from my PS2 I am able to use it normally with a PS4 and Xbox one X controller, but not with this PS3 controller.

grub3r-codes, 2024-01-02:
I am having a similar issue, with the same model ps3 sixaxis controller. Logs show that BR is not detecting the OEM controller as PS3:

`# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_CONN_REQUEST
# bt_hci_cmd_accept_conn_req
# Page dev: 0 type: 0 bdaddr: 00:26:5C:50:4A:19
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_ROLE_CHANGE
# BT_HCI_EVT_CONN_COMPLETE
# dev: 0 acl_handle: 0x0080
# bt_hci_cmd_le_set_adv_disable
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_STATUS
# BT_L2CAP_CONN_REQ
# BT_L2CAP_CONF_REQ
# BT_HCI_EVT_INQUIRY_COMPLETE
# bt_hci_cmd_exit_periodic_inquiry
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_passive
# bt_hci_cmd_le_set_scan_enable
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# dev: 0 error: 0x22
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# dev: 0 error: 0x02
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# dev: 0 error: 0x02
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_le_set_scan_param_active
# bt_hci_cmd_le_set_scan_enable
# bt_hci_cmd_periodic_inquiry
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_DISCONN_COMPLETE
# DISCONN for non existing device handle: 0080`

closed 2025-04-16 by darthcloud