darthcloud/BlueRetro #666: DS4 Controllers, Aftermarket, Bootleg, Fakes

Leerz, 2023-04-07:
### BlueRetro firmware version

v1.8.3_hw1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

espwroom

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

BOOTLEG PS4 Controller

### What is problem? (only list ONE problem per report)

Blueretro doesn't recognize bootleg ps4 controllers.
bootleg ps4 controller works on my ps4, pc, and android device. 
Originals work perfectly so I know my wiring is good for both p1 and p2

### What did you expect to happen?

bootleg ps4 controller works on my ps4, pc, and android device. 
i was kinda expecting blueretro would accept it as well

### Attach files like logs or Bluetooth traces here

n/a

retrofank, 2023-04-10:
I would not call this an issue (bug) imo.  Clone DS4 controllers use a different chipset, hence why it's a copy.   This not a BR problem and is problem with chinese bootlegs, as only legit Sony controllers were tested.

darthcloud, 2025-01-24:
I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

Thanks

Leerz, 2025-01-24:
> I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions? https://github.com/darthcloud/BlueRetro/wiki/Debug-trace
> 
> Thanks

this is acknowledged, will prep a setup this weekend. thank you for looking on to it.

darthcloud, 2025-04-16:
Feel to reopen once you got the trace

closed 2025-04-16 by darthcloud