darthcloud/BlueRetro #669: tracing bluetooth ds4 china

SpaceMyrzic, 2023-04-14:
### BlueRetro firmware version

1.8.3

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

BlueRetro Core

### System used

Sony PlayStation 1

### Bluetooth controller brand & name

CUH-ZCT2E

### What is problem? (only list ONE problem per report)

Can't pair Chinese DualShock4 with BlueRetro.

### What did you expect to happen?

Chinese DualShock4 connects well to PS4 and PC.

### Attach files like logs or Bluetooth traces here

Bluetooth HCI trace with Win10
[https://drive.google.com/file/d/1WwVeN_AqUlwXo3P1s6s6RvYaTNnI8VOp/view?usp=share_link](url)
[https://drive.google.com/file/d/1MXV0hOP_fj-Q_dHhL3wKxGSugdTq2Xee/view?usp=share_link](url)
BlueRetro debug logs via Serial port Windows 10
[china.txt](https://github.com/darthcloud/BlueRetro/files/11233605/china.txt)

dsErick, 2023-10-07:
I had ta same problem with that controller. @SpaceMyrzic have you found a way to pair with BlueRetro?

SpaceMyrzic, 2023-10-07:
On firmware 1.8.3 it was not possible to connect. I haven't tried the new firmware yet.

dsErick, 2023-10-08:
I tried 1.8.4 HW1 and HW2 and both I couldn't pair the ds4

darthcloud, 2025-01-24:
I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

Thanks

SpaceMyrzic, 2025-01-27:
I'll try to try it this week.

SpaceMyrzic, 2025-02-04:
Привет. Вроде всё сделал по инструкции.

darthcloud, 2025-02-04:
You need to upload the file here.

SpaceMyrzic, 2025-02-05:
[br_debug_trace.zip](https://github.com/user-attachments/files/18666748/br_debug_trace.zip)

darthcloud, 2025-05-06:
Can you try this build with it?

[v25.04-1-g83afb3b_hw1.zip](https://github.com/user-attachments/files/20024661/v25.04-1-g83afb3b_hw1.zip)

Make another trace if it doesn't work.

SpaceMyrzic, 2025-05-09:
[br_debug_trace (1).zip](https://github.com/user-attachments/files/20117363/br_debug_trace.1.zip)

SpaceMyrzic, 2025-05-09:
gamepad not connected