darthcloud/BlueRetro #672: Damaged N64 Controller Pak

Mailmansson, 2023-04-23:
### BlueRetro firmware version

[1.8.3] 2023-02-15

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

8bitmods

### System used

Nintendo 64

### Bluetooth controller brand & name

Nintendo Switch Online N64 controller

### What is problem? (only list ONE problem per report)

I cant get the controller pak to work. Every game i try it said "Controller pak may be damaged try to repair". I have factory reset the device multiple times. I have selected memory pack in the advanced config and still nothing fixed my problem. 

### What did you expect to happen?

to use the memory card to save

darthcloud, 2023-04-23:
The BlueRetro mem pak need to be first formatted to be usable. Not every games is able to do the format.
You can use the web config to perform the format:https://blueretro.io/n64_ctrlpak.html
Alternatively you can also use the Everdrive mem pak manager to do the format too.

Also are you using an N64 with an UltraHDMI or PixelFX N64 Digital?

They both have an HW issue that cause interference with the the first controller port and will make BlueRetro rumble & mem pak feature unusable.

You can get a patch flex cable from PixelFX or follow the intruction in this thread:
https://github.com/darthcloud/BlueRetro/issues/264#issuecomment-1214224340

Mailmansson, 2023-04-23:
Yes i am using a PixelFX N64 Digital. Looks like i need to order that new flex cable

closed 2023-04-23 by darthcloud