darthcloud/BlueRetro #677: PS2 Blueretro not compatible with Call of Duty 2 & 3

DOOTGUY, 2023-05-02:
### BlueRetro firmware version

1.8.3

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

SONY Dualshock 4

### What is problem? (only list ONE problem per report)

COD 2: Big Red One & COD 3 do not respond to interaction with the controller connected via BlueRetro to Playstation 2

### What did you expect to happen?

I expected the controller to work

sausehuhn, 2023-05-11:
This may or may not be correlated in any way, but I just had a similar experience with both WipEout Fusion and WipEout Pulse. The games wouldn’t react to controller input – the PS2 menu and another game would. Restarting the game with an Dualshock 2 plugged in, unplugging it and inserting the BlueRetro Core solved the issue. Worth a try maybe?

YgorSan, 2023-07-16:
maybe this is due to the pressure feedback of analog buttons on DualShock 2, all buttons except start, select, analog, R3 and L3 had this and currently blueretro does not support this function so there may be some incompatibility in games that require this, o which may be these cases. Alternatively, you can try using the pademu feature included in OPL, POPStarter and other homebrews for PS2 but unlike blueretro it has little compatibility and its use is limited to homebrews that support it (in current builds it supports PS3/PS4 and PS3 guitars , for some forks of modular pademu also support modules for wired xbox 360 controllers, xbox one, hid controllers, xpert and xbox 360 guitars, but they are not so polished forks compared to the main project in OPL). There are also brook adapters that maybe already support this...

DOOTGUY, 2023-07-25:
> maybe this is due to the pressure feedback of analog buttons on DualShock 2, all buttons except start, select, analog, R3 and L3 had this and currently blueretro does not support this function so [...]

I think that's not the problem, GTASA and Gran Turismo 4 work without problems, despite the fact that there is pressure feedback

jerzyjurgiel, 2023-09-29:
I also have issues in Yakuza 2 for PS2.
When I try to start the game with blue retro, console freezes. With wired DS2 it works no problem.

darthcloud, 2023-10-03:
@DOOTGUY  I fixed the problem with CoD 2, 3 & Kung-Fu Panda (maybe other Activision games too?)

You can try the v1.9 beta here:
https://github.com/darthcloud/BlueRetro/actions/runs/6386376011

For GT4 requiring to unplug/plug I think it's a OPL issue, I need to do that even with OG controller.
Running GT4 on an unmodded system BlueRetro work fine for me.

@sausehuhn Both Wipeout games work for me with BlueRetro v1.8.4 under OPL. Can you provided more detail? (PS2 model number, region, mods done to console, how to games is started)

@jerzyjurgiel Yakuza 2 work for me with BlueRetro v1.8.4 under OPL. Can you provided more detail? (PS2 model number, region, mods done to console, how to games is started)

jerzyjurgiel, 2023-10-03:
@darthcloud It's PS2 Slim model SCPH-77003, running newest stable OPL.
Game is being loaded via SMB, no extra modes configured.
I'm using pmgducati's PCB with 1.8.4 universal FW.
When playing the game with regular wired DS2, I have no problems and so far made it 2+ hours into game.
When blueretro is connected, game  freezes on loading or even in menu with and speakers emit constant buzz.
I discovered with padtest homebrew that with blueretro connected and single DS4 paired, it shows that pad 1 is recognized, but pad 2 is shown detected for split second and then dissapears. Maybe that's the issue ?

darthcloud, 2023-10-03:
@jerzyjurgiel try without plugin the p2 plug

jerzyjurgiel, 2023-10-05:
@darthcloud It works perfectly without P2 plug.
Will leaving it floating affect system detection or cause issues with Blue Retro?

darthcloud, 2023-10-15:
Fixed in v1.9 https://github.com/darthcloud/BlueRetro/releases/tag/v1.9

closed 2023-10-15 by darthcloud

djmtech, 2024-04-18:
Gran Turismo 4 doesn't work unless you unplug and plug the adapter back in. Will there be a fix for this in the future? Thanks

Espinete7, 2024-09-27:
Hi,

I have one problem with the BLUERETRO receiver, I can’t finde info about it.
I play ps1 and ps2 games in my ps2 original hardware. When I play with a wired dualshock 2 vibration works flawlessly, no matter if I use the discs or if I use pops starter for ps1, vibration works.
When I use the BLUERETRO receiver, with ps2 games I got the vibration, but when I play my ps1 games I don’t have any vibration in any of the games I tried.
What can I do to have vibration playing my ps1 games in my ps2?

Thanks in advance!