darthcloud/BlueRetro #686: [PS2] Cannot trigger IGR using controller connected via BlueRetro

PJSOConnell, 2023-05-13:
### BlueRetro firmware version

V1.8.3

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

Bought from BlueRetro store on AliExpress

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Xbox One S Controller

### What is problem? (only list ONE problem per report)

In-Game Reset (IGR) cannot be triggered by controllers connected to the BlueRetro dongle. 

Steps To Reproduce -
Using a modded PS2 (hard/soft mod), open OPL and run a PS2 game. 
When the game has loaded, enter the IGR button combination - L1+L2+R1+R2+Select+Start

### What did you expect to happen?

I expect to be booted back to the OPL screen. 

Actual Behaviour - nothing occurs. The button combination works as normal with a wired controller.

YgorSan, 2023-07-16:
in this case I think it is not an incompatibility related to blueretro but to the IGR feature itself, to begin with it is not compatible with all games and cannot have OPL compatibility mode 6 active. I use the function in both OPL and POPStarter with a DualShock 4 connected (it also worked with other bluetooth controllers for android), not to mention that my blueretro is the same as yours and with the latest firmware installed. An example of a game that doesn't work well with this is Area 51 but curiously works well with the other IGR function to turn off the console (pressing the L1+L2+R1+R2+L3+R3 buttons).

diegoadx, 2023-09-20:
I am having the same issue, IGR won't work with my DS4 controller. But it does register the input combination with Pad Test.

jerzyjurgiel, 2023-09-25:
I can confirm, same issue for me. IGR doesn't work on ps2 with DS4

darthcloud, 2025-05-13:
OPL IGR work for me while using BlueRetro, example Resident Evil 4.

I use the "lastest" release they made back in 2021:
sha1sum OPNPS2LD-v1.1.0.ELF 
be594ff83a81486c20b3f7276bcef3041bea29f3  OPNPS2LD-v1.1.0.ELF

I can't get the latest betas of OPL to work on my FHDB setup, they just give me a black screen.

closed 2025-05-13 by darthcloud