darthcloud/BlueRetro #688: (Gamecube Internal) 8BitDo Pro 2 Only Syncs To Second Controller Port

viggie24, 2023-05-14:
BlueRetro firmware version

1.8.2

BlueRetro firmware specification

HW2

BlueRetro firmware variant

System specific

BlueRetro hardware type

Internal install modkit

Manufacturer

Laserbear

System used

Nintendo GameCube

Bluetooth controller brand & name

8bitdo Pro 2 Bluetooth Controller

What is problem? (only list ONE problem per report)

Controller always syncs to the second controller port even when the first port is open on X input. 

What did you expect to happen?

The controller should sync to port one. 

Attach files like logs or Bluetooth traces here

No response

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2023-05-20 by viggie24

darthcloud, 2023-05-20:
What fixed your issue?

viggie24, 2023-05-21:
I think it was an issue on 8bitdo’s end. There was a new firmware update for the controller and now it connects perfectly. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>