darthcloud/BlueRetro #690: HW2 N64 Ports Always Active

jtauby, 2023-05-17:
### BlueRetro firmware version

1.8.2 / 1.8.3 HW2

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

I built all HW other than current mirror boards from Helder.

### System used

Nintendo 64

### Bluetooth controller brand & name

OEM PS4 Controller

### What is problem? (only list ONE problem per report)

When I turn the console on and the ESP32 boots up, it looks like there are always (4) controllers plugged into the system. I use Super Smash Brothers as my test game because it lets you easily visualize the connected controllers. When I get into the game, it shows all (4) controllers as active with human users, even if I have no BT controller or wired controller plugged in. I can connect all (4) BT controllers to the BlueRetro fine, and the data from the controller is passed correctly. If I plug a wired controller into any of the ports, it still passes the data and it correctly bumps the downstream BT controllers to the next port.

If I check the voltage at the sense pin of the current mirror boards, I am getting the correct signals (3.3V for open, 0 for controller plugged in). 

Is this the correct functionality? I assumed that the if there were no BT controllers or wired controllers connected, that the system would show those controller ports as unused.

Thanks for your help, this is a great project. 

### What did you expect to happen?

I expected the ports to be "off" if there is no BT controller or wired controller connected to that specific port.

darthcloud, 2023-05-17:
N64 controller are technically not hot plugable and as such BlueRetro is always active on the port, unless a wired controller is detected on HW2.

So this is proper behavior, its done this way cause the vast majority of games only detect controller inserted at boot.

closed 2023-05-17 by darthcloud

jtauby, 2023-05-18:
thanks for the response! Everything is working as intended then.