darthcloud/BlueRetro #692: Dreamcast keyboard no response

rpf16rj, 2023-05-18:
Hi, how are you? I hope fine!

I am try use keyboard on a Dreamcast. I made a adapter and he work fine with a bluetooth PS4 joystick.
When i try use in keyboard mode (i setted the mode to Keyboard and preset Gamepad;Keyboard), the kayboard show connected on Dreamcast but no response of keys. 

I tested that this version of fw in this thread and the lasted 1.8.3.

In another devices, the both keyboards works fine.

I tested with two models of keyboards bt, a Microsoft Mobile Keyboard 6000 and a generic bt keyboard chinese. I atached the debug log below. 

Thanks for you time. 

[bt_keyboard_microsoft_mobile_6000.txt](https://github.com/darthcloud/BlueRetro/files/11479638/bt_keyboard_microsoft_mobile_6000.txt)
[bt_keyboard_chinese.txt](https://github.com/darthcloud/BlueRetro/files/11479639/bt_keyboard_chinese.txt)

_Originally posted by @rpf16rj in https://github.com/darthcloud/BlueRetro/discussions/332#discussioncomment-5907052_

darthcloud, 2023-10-09:
Get you get a BT trace like explained here:
https://github.com/darthcloud/BlueRetro/wiki/Bluetooth-HCI-trace-with-Win10

for both

Thanks!

rpf16rj, 2023-10-09:
Follow, thanks.

[bt_tracing.zip](https://github.com/darthcloud/BlueRetro/files/12846981/bt_tracing.zip)

darthcloud, 2024-04-22:
From the trace of the chinese KB:

\# DBG handle: 0 dev: 0 type: 0
\# dev: 0 type: 0:0 Bluetooth Keyboard
\# adapter_debug_task HID descriptor
\# 000000 05 01 09 06 A1 01 85 01 75 01 95 08 05 07 19 E0
\# 000010 29 E7 15 00 25 01 81 02 95 01 75 08 81 03 95 05
\# 000020 75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 03
\# 000030 95 06 75 08 15 00 26 FF 00 05 07 19 00 29 FF 81
\# 000040 00 C0 05 0C 09 01 A1 01 85 02 15 00 26 3C 04 19
\# 000050 00 2A 3C 04 75 10 95 01 81 00 C0 05 01 09 80 A1
\# 000060 01 85 03 25 01 15 00 75 01 09 81 09 82 09 83 95
\# 000070 03 81 06 95 05 81 01 C0 05 0C 09 01 A1 01 85 11
\# 000080 15 00 25 01 75 01 95 03 81 01 75 01 95 01 05 0C
\# 000090 09 B8 81 02 06 FF 00 09 03 81 02 75 01 95 03 81
\# 1 I 07E0 0 8 0700 16 8 0700 24 8 0700 32 8 0700 40 8 0700 48 8 0700 56 8 rtype: 0
\# 1 O 0801 0 5 
\# 2 I 0C00 0 16 rtype: 3
\# 3 I 0181 0 3 
\# 17 I 0CB8 3 1 rtype: 3
R0 07E0: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 0F, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 18, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 08, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 05, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 0F, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 18, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 08, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 28, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
R0 07E0: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, R0 0700: 00, 
\# DBG DISCONN from handle: 0 dev: 0

HID parser look to be working for that one.
Issue look to be in the handling of fragmented packets used for the HID descriptor.
I will have to investigate further.

I dont have the trace for the MS 6000 KB but from the log this look to be a HID parser error:
# 1 # Unknown HID marker: 1A

darthcloud, 2024-04-22:
Can you try this beta with both the Microsoft 6000 and the Chinese.

https://github.com/darthcloud/BlueRetro/actions/runs/8790166432

This should fix the 2 issues i saw in the log/trace

rpf16rj, 2024-04-23:
> Can you try this beta with both the Microsoft 6000 and the Chinese.
> 
> https://github.com/darthcloud/BlueRetro/actions/runs/8790166432
> 
> This should fix the 2 issues i saw in the log/trace

Hi, i can confirm that MS 6000 work now. 
Chinese doesn't work.

darthcloud, 2024-04-23:
Can you get the serial console again on that beta while connecting the Chinese one?

darthcloud, 2024-04-23:
Thanks for the log, that keyboard descriptor is simply very unusually big. I doubled the size available and it still dont fit.

It won't be possible to support it in BlueRetro unfortunately.

Thanks for providing the logs and sorry for long delay.
At least the MS one work now.

darthcloud, 2024-04-24:
Fixed in v24.04