darthcloud/BlueRetro #706: 8bitmod n64 receiver

0bie0ne, 2023-05-31:
### BlueRetro firmware version

1.8.3

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

8bitmod

### System used

Nintendo 64

### Bluetooth controller brand & name

8bitdo n64 controller

### What is problem? (only list ONE problem per report)

i have to re-sync controller because A button is B and B is A button.

### What did you expect to happen?

A is A B is B without syncing again

v1605, 2023-06-02:
Also experienced this issue with the controller is switch mode. Switching to windows/android cause B to register as down c.

Traxus7, 2023-06-07:
> Also experienced this issue with the controller is switch mode. Switching to windows/android cause B to register as down c.

Same issues here, 1.83 firmware, on a Funtastic Grape N64.

Pak set to "S" mode: B and A are swapped
Pak set to "D" mode: B registers as Down-C and Z is unmapped

Beer-Fridge, 2023-06-09:
Same issue for me as well. Anytime the controller reconnects after the initial sync, the A and B buttons are swapped.

nortakales, 2023-06-09:
This probably warrants a separate issue (or maybe none at all since it may not be something BlueRetro can even change/fix) but rumble is not working with the 8bitdo N64 mod kit. I've gone back and forth with 8bitdo support, and they claim it only works with Nintendo Switch, and only NSO N64 games. Which is a serious shame. I would LOVE to have rumble working on my real N64 through BlueRetro and the 8bitdo N64 mod kit. That kind of seems like the holy grail of playing N64 in 2023.

Traxus7, 2023-06-10:
> This probably warrants a separate issue (or maybe none at all since it may not be something BlueRetro can even change/fix) but rumble is not working with the 8bitdo N64 mod kit. I've gone back and [...]

So apparently the 8bitdo support response is their rumble currently only works on Switch NSO games. Disappointing, and seems like a strange (unfinished) design choice to only make it fully-functional specifically on Switch (for specific Switch NSO games) while not enabling rumble at all for D-input windows/android.

"Greetings, 
Thank you for the consultation.
Please note that ONLY the N64 games on Nintendo Switch(8BitDo N64 Mod Kit in S mode) that have the rumble support can achieve the rumble feature on our N64 Mod Kit. There will not have rumble feature on this N64 Mod Kit with any other games, please see the attachments in this email for more information.

    The 8BitDo N64 Mod Kit cannot be used on an original N64 console.

    The 8BitDo N64 Mod Kit does not support rumble feature in Dinput working mode.

    We do not guarantee our N64 Mod Kit would work properly without issue with the "BlueRetro Bluetooth Receiver" since it is not from 8BitDo."

Traxus7, 2023-06-12:
Okay, I have a temporary mapping workaround to avoid re-pairing every time. I assume on re-pairing it's being detected as some kind of basic Switch controller but the broken mappings don't seem to match up with any controller 1:1.

With these changes in Advanced Config, on subsequent connections after the initial pairing of an 8bitdo modkit in "S" mode, all buttons are correctly mapped with the exception of C-Right, which is unresponsive, but I've mapped that function to the ZR button on the pak itself.

On re-pairings the ZR button on the pak is mapped as another C-Down normally, but I've remapped it as C-Right since I'm at a loss as to which "Src" cell applies to the actual unresponsive C-Right button. If anyone can find it, please let me know. I've literally tried changing every single Dest cell to C-Right, and every button on my controller spins the camera to the right in SM64 EXCEPT for C-Right, which remains nonfunctional.

[image]
Just set the labels easy viewing, change the cells under Dest highlighted in green to appear as shown, scroll down and save.

I hope this helps until darthcloud gets a chance to look at the modkit later this summer!

P.S. The 8bitdo boards in "S" mode do NOT suffer from the latency or drops of issue #146 like other Switch controllers. I can have all 4 of my 8bitdo controllers paired in "S" mode and they all behave normally.

DS20815, 2023-06-30:
I am experiencing the same issue with a combination of the 8bitmod N64 receiver and the 8bitdo n64 mod kit in S-Mode. Very rarely the controller pairs fine and all buttons are working as intended but in 99% of the time the button mapping is off; A=B, B=A, C-Up=C-down, C-Down=Z and C-Right is not working at all. I already thried the suggested remapping but I am not able to get C-Down working (except for mapping it to ZL, which is unplayable for me). 

Sometimes - very rarely - the controller also pairs but does not recognize any input.

I also tried D-Mode but there the Z button seems to be unavailable.

I am not quite sure wheter this is an error in the 8bitdo firmware or a pairing problem between the controller and the blueretro receiver. Overall the mod kit and the receiver are great and could in combination be the best wireless solution for the N64, with a far better availability than the NSO controller. The controller is working perfectly fine with the switch, but due to the mapping issues with blueretro the kit is no alternative to wired controllers at the moment.

faha223, 2023-07-30:
I may have found something useful. 

I've only tested this a handful of times, but it seems that if you put the N64 controller into pairing mode before turning the N64 on (press and hold the sync button), then it syncs up and works correctly. If you just leave it in the slow flashing mode then it'll eventually connect but the button bindings will be all messed up.

This is the workaround I'll be using from now on as it seems to be reliable (according to my roughly 3 minutes of testing).

Traxus7, 2023-07-31:
> I may have found something useful.
>
> I've only tested this a handful of times, but it seems that if you put the N64 controller into pairing mode before turning the N64 on (press and hold the sync [...]

On first pairing the mappings are correct, so yeah this is a solid workaround for a single controller, or controllers each paired with individual dongles. 

It's just unwieldy if you're pairing multiple controllers on a single multi-cable blueretro board, since after the first controller pairs the receiver leaves pairing mode and only monitors for controllers that have been paired previously (I don't know what the technical term is). I guess it's possible to re-enter pairing mode by pressing one of the buttons on the unit, but I haven't tried it. It's possible that would require the first controller to re-pair as well? Which would make the whole workaround pointless in this situation.

darthcloud, 2023-10-07:
I added support for the **D mode** (switch mode is broken)

You can try v1.9 beta here:
https://github.com/darthcloud/BlueRetro/actions/runs/6437347093

See pairing instruction in the beta wiki:
https://github.com/darthcloud/BlueRetroWiki/blob/master/Controller-pairing-guide.md#15---8bitdo-n64-modkit-paring-guide

stainmage, 2023-10-07:
thank you very much darthcloud, i'll try it right now, hope the switch mode will get fixed so we can enjoy the rumble action !

nortakales, 2023-10-07:
@darthcloud is it even possible to get Switch mode working eventually, or will it never be possible? Just curious.

stainmage, 2023-10-07:
hey this is the workaround in D mode

[image]

darthcloud, 2023-10-07:
You shouldn't need any work around with latest version. try to factory reset your adapter

darthcloud, 2023-10-07:
@nortakales official Switch is a giant PITA, 3rd party ones even more so.

Likely not, complain to 8bitdo that rumble don't work in D mode.

stainmage, 2023-10-07:
did this all the day but it didn't work without this darthcloud, not sure if the 1.9 is linked above, it downloads as 1.8.4 and it needs remap even after factory reset

darthcloud, 2023-10-07:
the beta will display as v1.8.4-27 in web

darthcloud, 2023-10-07:
note that factory will remove the update, so you need to uodate again after

stainmage, 2023-10-07:
i'll give it a try again thanks

darthcloud, 2023-10-07:
someone else confirmed it work in the other bug.

Did you ever update the 8bitdo controller FW?

I'm on stock one.

stainmage, 2023-10-07:
the only fw available is the 1.0 afaik

stainmage, 2023-10-07:
did a factory reset then reflashed the firmware, but it didn't work, mapping is still messed on my side, need the workaround again

0bie0ne, 2023-10-07:
seem to be working no issue, power off console and controller. turn on console then controller re-sync automatically all buttons works

stainmage, 2023-10-08:
maybe i'm flashing it wrong ? i'm using the retroscaler blueretro-n64 from the bluetooth web interface, should I use hw1 or hw2 ?

darthcloud, 2023-10-08:
HW1, its the proper way. Can you check your controller FW version in the 8bitdo software? I'll try to check mine later.

0bie0ne, 2023-10-08:
8bitdo has only one fw 1.00.

darthcloud, 2023-10-08:
@stainmage 

Mine got V1.00, no idea why your buttons would be reversed. The controller rumble switch is really in D mode? And you repaired while being in that mode?

What version do the BlueRetro web interface show? Take screenshot.

Can you get a trace of your controller while in D mode on windows?
https://github.com/darthcloud/BlueRetro/wiki/Bluetooth-HCI-trace-with-Win10

While doing the trace first press button A like 10 times and then after press B buttons 10 times. Then stop the trace, don

solid1986, 2023-10-09:
Updated firmware on my adapter to v1.8.4-27-g0a29947_hw1 and I'm also having the same problem. At first I thought everything was working, so I moved on to my next problem, n64 digital was causing my screen to tear occasionally. I was testing that fix with smash brothers and noticed the B button caused me to jump in the air, rather than the normal moves. Can't navigate back in menu's in the everdrive. Didn't look at button mapping directly, but it's certainly not working as intended. Connects like a champ now without the previous bugs though.

darthcloud, 2023-10-09:
Can you go load the default preset via the beta web config to be sure you don't have old mapping you previously made

https://darthcloud.github.io/BlueRetroWebCfg_Beta/presets.html
Select default gamepad
then save

stainmage, 2023-10-10:
Hi darthcloud, the controller is in D mode, was previously linked in switch mode but I reset everything so it's not paired anymore.

here's a screenshot of the web interface

I didn't try to shutdown the n64 after factory reset and after firmware update, i'll try again this way, and will load the default gamepad via web config beta and let you know asap.

Will do the trace if it doesn't work and let you know

[image]

stainmage, 2023-10-10:
everything works fine now ! thanks

darthcloud, 2023-10-10:
Base on screeshot the FW ersion was wrong.

Should be:
```
Connected to: BlueRetro_N64_xxxx (xx:xx:xx:xx:xx:xx) [v1.8.4-27-g0a29947 hw1]
```

stainmage, 2023-10-10:
it's fine when I check the version before starting to flash, it shows Connected to: BlueRetro_N64_xxxx [v1.8.4-27-g0a29947 hw1 ] but it shows this suggestion to download 1.8.4 from itch.io when it starts flasing

darthcloud, 2023-10-10:
it's normal, it's a beta firmware so just ignore that message

solid1986, 2023-10-10:
I resolved my problem. Apparently I still wasn't on the correct firmware version, even though I definitely downloaded and unzipped the latest. Not sure what happened there, but I re-downloaded and installed again and everything is working flawlessly. Thanks for everything you do!

Traxus7, 2023-10-13:
OMG THANK YOU DARTHCLOUD!! Finally, got my four modded original N64 controllers on a single BlueRetro board with no weird workarounds. The dream is a reality.

That's a shame about no rumble but it makes zero sense that 8bitdo didn't implement it at all in D mode from day-one. I think I'll write 8bitdo an email a week about it until they update that shit.

Matt-Retrogamer, 2023-10-13:
Just to understand properly, the rumble pack does not work at all with the BlueRetro adapter and the 8bitdo mod kit ? It makes no vibrations neither in D nor S modes? Just works on Switch in NSO?

nortakales, 2023-10-13:
@Matt-Retrogamer that seems to be the current state, yes.

I emailed 8BitDo when the adapter first came out. It was really tough communicating with them because I only know English and English was not their first language, but from what I could gather 8BitDo says this is totally expected behavior and had no plans to support rumble on anything other than NSO. Which is an utter shame, IMO.

darthcloud, 2023-10-13:
BTW the rumble not working in Switch mode is not unique to BlueRetro. All 8bitdo controller used with steam on windows will not have working rumble after reconnect (not sure if it work after a first pairing)

I did actually investigate this in detail going all the way to do a wireless trace of a 8bitdo controller with a real Nintendo switch. BlueRetro perform the config just like a real Switch would. But for some unknown reason the rumble packets are ignored by the 8bitdo controller 🤷.

Traxus7, 2023-10-13:
> I emailed 8BitDo when the mod kit first came out. It was really tough communicating with them because I only know English and English was not their first language, but from what I could gather 8BitDo says this is totally expected behavior and had no plans to support rumble on anything other than NSO. Which is an utter shame, IMO.

Agreed, though plans can easily change if enough people bug them about it. I wouldn't assume it's a hardware impossibility. When people asked about rumble back at launch their canned response was it's "currently" only available in Switch mode.

AFAIK they have other Switch / Android / PC compatible controllers that do allow access to rumble functionality in all modes? Kinda feels like they just rushed the N64 modkit out as soon as they had the Switch side of things working and left D-input partially functional.
[image]

darthcloud, 2023-10-15:
Fixed in v1.9 https://github.com/darthcloud/BlueRetro/releases/tag/v1.9

closed 2023-10-15 by darthcloud

Matt-Retrogamer, 2023-10-19:
Sure... But what about S mode ?

Traxus7, 2023-10-19:
> I emailed 8-bitdo. Here is my request:
> 
> _I am writing in hopes that you can update your N64 Controller Modkit with Rumble support in D Mode. It would be a huge selling point for people that use their controllers on devices other than the Nintendo Switch._
> 
> Here is their response:
> 
> _Thank you for reaching us out!_
> 
> _Dinput mode is a protocol that does not have vibration capability by default, and even if vibration functionality is added, the device connected to the controller still won't recognize the vibration._

Major bummer. I got the same copy/paste response, with the additional claim that none of their controllers support rumble for D-input:

_Thank you for reaching us out!_
_All of the 8BitDo controllers do not have the rumble feature in Dinput mode. Dinput mode is a protocol that does not have vibration capability by default, and even if vibration functionality is added, the device connected to the controller cannot recognize the vibration.

Kind regards,
-----------------
Max
8BitDo Support_

dragonblue950, 2023-10-30:
Thank you stainmage for the tip, you save me. I use a blueretro found on Aliexpress and I can't install v1.9 it tells me it's incompatible. I must use the NodeMCU-PyFlasher software and a link given to update it but the latest version is n64_1.8.3.bin. Sorry for the Google translate.

[image]

https://github.com/RetroScaler/N64-Blueretro/

https://github.com/RetroScaler/N64-Blueretro/releases

drackybrok, 2024-04-07:
> Sure... But what about S mode ?

When I had a back and forth with support for rumble in s mode for platforms other than switch online they said they had no plans to add any.