darthcloud/BlueRetro #720: Dreamcast internal ESP32 install has trouble connecting on startup

Tronicgg, 2023-06-22:
### BlueRetro firmware version

1.8.3

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

ESP32

### System used

SEGA Dreamcast

### Bluetooth controller brand & name

DualShock4, Switch Pro Controller

### What is problem? (only list ONE problem per report)

Having trouble connecting to DS4 & Pro Controller on system startup. Generally it won't connect to synced controllers on startup. I have to reset ESP32 (installed an external switch) every time for successful connection. Random disconnect every now and then too. Any ideas on where to start troubleshooting?

### What did you expect to happen?

Normal sync.

konwektor, 2023-08-27:
I got simillar issue in gamecube internal install with xbox controllers. If I use combo "Disconnect controller (+ power off internal mod)", esp32 is disconnecting all controllers, switching off console, and waits for BT connections from paired gamepads.(because still got power). This way if i switch on gamepad, it connects quite fast to blueretro. Other way if i use boot btn on esp32 to disconnect paired controllers, reconnecting sometimes takes a long, long time, or doesnt work at all. If i switch manually console off, with pwr button, same problems with reconnecting, long time, and mostly if reconnected, combo doesnt work anymore (all combos) - in console log I see that combo has been pushed, but doesnt do anything, no reaction.
Best solutions in my case - use combo to power off console.

vsilvalopes, 2023-11-23:
Hi!
I've got a similar problem, and in my case was a faulty ESP32 module. 

I've followed this wiring here : [https://blogger.googleusercontent.com/img/a/AVvXsEhEEHrUV8xf8Lb1pUbTi-9clwhY9pi5wiNO3tkd9Rq3EPIOHbxG524GJ0dFGXF24Vt8d_aRD_s8yPBsJMFJHzm3L4vQd-HqYb1DwDlonZeq9mwaMqoYREwrZufBlq2Yxdc4oKemPoERgIFZPLnG7fwhUCUEbUgk-hotoaqyMaP3vLMPuy5Puzvc2k8vdA=s3300](url)

It's from a Colombian modder, Yakara.

Check it out, because mine is working as expected, but at first I've got the same problem as you. And it was a faulty ESP module.

Tronicgg, 2023-11-23:
Thanks mate, will try swapping it for a new module.

Borja
Lead Designer
BADV Studio

+34 625 880 262
www.badvstudio.com

darthcloud, 2024-04-20:
@Tronicgg did swapping module resolve your issue?

Tronicgg, 2024-04-21:
> @Tronicgg did swapping module resolve your issue?

This one has not completely gone away. It does work most of the time, but still have to power cycle the ESP32 because it wont connect on startup sometimes. Before changing the module I had to power cycle on every Dreamcast startup. Now only sometimes.

VMU is working fine after a couple weeks of testing.

darthcloud, 2024-10-06:
Likely something related to the DIY internal install rather than FW itself. Sound like it could be a power related issue.

Closing not much I can do.

closed 2024-10-06 by darthcloud