darthcloud/BlueRetro #732: Turbo does not function correctly on Neo Geo platforms when using an 8BitDo controller

DoomyDoomer, 2023-07-18:
### BlueRetro firmware version

1.8.3

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Humble Bazooka

### System used

SNK NeoGeo AES

### Bluetooth controller brand & name

8BitDo M30, 8BitDo MD DIY, 8BitDo Saturn DIY

### What is problem? (only list ONE problem per report)

When using 8BitDo controllers (possibly others), the Turbo feature does not work as expected on the Neo Geo AES or MVS. Pressing a button setup with turbo does not activate the turbo functionality. In some cases, pressing a button setup for turbo activates other buttons on the controller even when they aren't pressed. For example, in Burning Fight pressing "C" (setup for turbo) activates "B" and "C" together every few seconds.

### What did you expect to happen?

The expectation is for any button setup for turbo in the webconfig to function as a turbo button when pressed.

darthcloud, 2023-12-09:
Fix is now in v1.9.1

closed 2023-12-09 by darthcloud

Metro-City, 2025-03-23:
With 2 dongles the machine resets, even after the update if you select a character it resets, only by removing the 2nd dongle can you play without the machine restarting.