darthcloud/BlueRetro #745: Cannot connect to the web interface.

dystopia2017, 2023-08-12:
### BlueRetro firmware version

v1.8.3_hw1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

ESPRESSIF

### System used

Nintendo GameCube

### Bluetooth controller brand & name

ESP32-WROOM-32

### What is problem? (only list ONE problem per report)

I can not connect to the web interface from Chrome.

`Live Output
Requesting Bluetooth Device...
Connecting to GATT Server...
Device name: BlueRetro_GC_4AA6
Device id: wkEROuuB3akr2GVv6ayhpg==
NetworkError
Argh! Couldn't connect to BlueRetro`

### What did you expect to happen?

I expected to be able to connect

### Attach files like logs or Bluetooth traces here

`Live Output
Requesting Bluetooth Device...
Connecting to GATT Server...
Device name: BlueRetro_GC_4AA6
Device id: wkEROuuB3akr2GVv6ayhpg==
NetworkError
Argh! Couldn't connect to BlueRetro`

DoomyDoomer, 2023-09-18:
This is fairly common. The best advice is to close the tab, open a new tab, and retry. Repeat this process until it finally connects. v1.8.4 seems to also connect more often than previous versions.

darthcloud, 2024-10-06:
I realise this is a chicken and egg problem.

But if you were able to update to v24.10 things would be much better. I increased the connection timeout in that version.

Your best bet for a successful update is to use a Win10 laptop with built-in Intel bluetooth close to the adapter location.

closed 2024-10-06 by darthcloud