darthcloud/BlueRetro #748: Can't use 2 DualShock 4s

SavnoorSamra, 2023-08-18:
### BlueRetro firmware version

1.8.4

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

Myself

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Sony DualShock 4

### What is problem? (only list ONE problem per report)

I am unable to connect two dualshock 4 controllers simultaneously. If I have a Dualshock 4 connected as player 1, any controller connected as player 2 will lag, have random button presses, and eventually disconnect and power off. If I have a different controller as player 1 and a Dualshock 4 as player 2, then everything works as expected.

I can use two DualShock 3s and two WiiMotes without problems but not 2 DualShock 4 controllers.

I couldn't seem to find anything in the logs.

### What did you expect to happen?

Both controllers would work without issues.

darthcloud, 2023-08-19:
Do you have that issue with 1.8.3? or it's only with 1.8.4?

closed 2023-08-22 by darthcloud

SavnoorSamra, 2023-08-24:
Hey I didn't get an email for your response... I never tried 1.8.3 since I just built the adapter so I can't say for sure.