darthcloud/BlueRetro #755: my control locks almost always

BragnarXIX, 2023-08-23:
### BlueRetro firmware version

1.8.4

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

Internal install DIY

### Manufacturer

No

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Xbox One Controller

### What is problem? (only list ONE problem per report)

Every time I play the controller it doesn't disconnect but it stays stuck and I can't do anything I have to turn it off and on again to make it work again like this repeatedly.

### What did you expect to happen?

Solution

BragnarXIX, 2023-08-23:
Sistem is esp32

darthcloud, 2025-04-16:
Feel free reopen once you got the trace.

closed 2025-04-16 by darthcloud