darthcloud/BlueRetro #759: Exclene gamecube controller instant disconnect v1.8.4

DragonKumera, 2023-08-28:
### BlueRetro firmware version

1.8.4

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install modkit

### Manufacturer

Lazer Bear

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Excelene Gamecube controller

### What is problem? (only list ONE problem per report)

Controllers constantly disconnecting after not even a minute. (exclene gamecube controller) 4 of them. Had to flash back to 1.8.3 for now as I tried all sorts of fixes or troubleshooting and nothing worked. Bright side rumble worked but I truly hope this gets fixed soon.

### What did you expect to happen?

Not disconnect

darthcloud, 2023-08-28:
Did you pair controller in Y + Home mode?
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#12---exlene-gamecube-paring-guide

DragonKumera, 2023-08-29:
Yes and same problem.

darthcloud, 2023-10-07:
Tried with my own Exlene in  v1.8.4 and it totally work fine for me in IOS/Xinput mode (Y + Home). Played 15min with no issue.

The fact you say rumble work tells me you are not connected in the right mode.

Also you may want to try latest v1.9 beta:
https://github.com/darthcloud/BlueRetro/actions/runs/6437347093

DragonKumera, 2023-10-07:
> Tried with my own Exlene in v1.8.4 and it totally work fine for me in IOS/Xinput mode (Y + Home). Played 15min with no issue.
>
> The fact you say rumble work tells me you are not connected in the [...]

I tested both the home + y pairing mode (that has no rumble) and just the home pairing mode that has rumble. It works fine if in use I believe on 1 controller. But when 4 are connected and left idle for a minute or less it loses connection.

I will try this beta version you suggested and see how it goes. Not sure if I will get to it this weekend but will try. I can also record a video if you want if the issue persists.

I know 1.8.3 has been working solid for me on both just pressing home to pair and home + y. But will report back once I do testing when I have a chance.

DragonKumera, 2023-10-08:
> Tried with my own Exlene in v1.8.4 and it totally work fine for me in IOS/Xinput mode (Y + Home). Played 15min with no issue.
>
> The fact you say rumble work tells me you are not connected in the [...]

Flashed the beta you linked and my controller disconnects were fixed! Now I am back to solid! The only issue now is the l+r button kinda half works? Its like it works but then sometimes doesnt or depending on the game doesnt at all. (mario party 7 it worked perfectly fine in). Well the L shoulder button. The R I had to kinda press harder? or press multiple times. Tested multiple controllers.

darthcloud, 2023-10-15:
Fixed in v1.9 https://github.com/darthcloud/BlueRetro/releases/tag/v1.9

closed 2023-10-15 by darthcloud