darthcloud/BlueRetro #761: controller (8bitdo arcade stick) not in the list

berke385, 2023-08-30:
### BlueRetro firmware version

1.8.4

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install modkit

### Manufacturer

Laser Bear Industries

### System used

Nintendo GameCube

### Bluetooth controller brand & name

8bitdo arcade stick

### What is problem? (only list ONE problem per report)

my controller is not in the list so when for example I press "x" on my arcade stick its recognized as a "b" or "y" and "b" are recognized as select.

### What did you expect to happen?

that the inputs are correct but after I looked up the controller list I noticed that the arcade stick is not in the list. so I installed wireshark for the bluetooth trace. I had to zip the file because github do not allowed me to upload the .pcapng file.

### Attach files like logs or Bluetooth traces here

I pressed x, a, y, b, rb, rt, lb, lt, p1, p2, start and select in this order (I pressed start and select in the last 2-3 times not at beginning)
[bl_trace_8bitdo_arcade_stick.zip](https://github.com/darthcloud/BlueRetro/files/12474876/bl_trace_8bitdo_arcade_stick.zip)

darthcloud, 2023-10-09:
Can you post a picture of the controller.

I look up online and button look to be label with Switch name.
But you use Xbox name.

berke385, 2023-10-22:
[image]
[image]
[image]

the controller has 2 options (for the nintendo switch and for windows etc.) and I had it on the "x-input" as in the last picture

Raton-Laveur, 2024-01-17:
Looks like @berke385 forgot Home (MT), Capture/Star (MQ) and the axes, so I'm putting a capture with them added in. 
Specific features for this device: 

- P1 and P2 are Macro keys, programmable through 8bitdo software - they may not appear in the trace. 
- Both sticks and Dpad are mapped and can be selected through a switch on top left (see pictures on previous post).

Order in which keys were pressed: 

- x, a, y, b, rb, rt, lb, lt, p1, p2, start, select, Home, Capture
- Left Stick: Up Down Left Right
- DPad: Up Down Left Right
- Right Stick: Up Down Left Right

[8bitdo-arcadestick.zip](https://github.com/darthcloud/BlueRetro/files/13969282/8bitdo-arcadestick.zip)

Raton-Laveur, 2024-11-19:
Good news: this controller is usable with some config. 

When paired with BlueRetro (24.10), it is recognized and usable, but buttons are in a unusable disposition - expected, as there is no arcade stick currently listed in BlueRetro.

For example, with "MD / Genesis", what it currently is / what it should be (reproducing the [Sega Mega Drive Arcade Power Stick II](https://www.starboardgames.com.au/product/sega-mega-drive-arcade-power-stick-ii-2-6-button-joystick-controller/)):

-GP:RB Left / GP:LS
-GP:RB Down / GP:RB Left
-GP:RB Right / GP:RB Down
-GP:RM / GP:RB Right
-GP:LS / GP:MS (emulating that the Mode Button is on the right shoulder of the Mega Drive 6-button pad)
-GP:LM / Nothing

**Workaround**

Using 8bitdo Ultimate Software, remap buttons. Here is a screenshot to the config I am proposing: 

[image]

darthcloud, 2025-01-24:
I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

Thanks

Raton-Laveur, 2025-01-24:
Sure, here it is - buttons pushed in the same order than [previous trace](https://github.com/darthcloud/BlueRetro/issues/761#issuecomment-1897100346).

[br_debug_trace-8bitdo arcade stick.zip](https://github.com/user-attachments/files/18533481/br_debug_trace-8bitdo.arcade.stick.zip)

Raton-Laveur, 2025-01-24:
Sure, here it is - buttons pressed in the same order than [previous trace](https://github.com/darthcloud/BlueRetro/issues/761#issuecomment-1897100346).

[br_debug_trace-8bitdo arcade stick.zip](https://github.com/user-attachments/files/18533701/br_debug_trace-8bitdo.arcade.stick.zip)

darthcloud, 2025-01-25:
@Raton-Laveur for some reason your trace looks wrong. I got a fews trace from others and they all look fine.

Can you try to make it again? Make sure you got the latest v25.01 and not an older beta. Make sure power cycle adapter after enabling debug mode.

Raton-Laveur, 2025-01-25:
I think I know what went wrong (*) - here are two more with HW1 powered by USB port, not connected to any console. 

[br_debug_trace-8bitdo arcade stick-2.zip](https://github.com/user-attachments/files/18545911/br_debug_trace-8bitdo.arcade.stick-2.zip)

Thanks!

(*) Yesterday's trace was made with HW1 powered by a Sega Mega Drive. I think pressing GP:MT (Home) [switched to GamePadAlt](https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-System-Specific-User-Manual#genesis) and made the trace go wrong. So today's traces are made just with HW1 powered through USB with no console attached. 
--> If this is proven right, documentation should reflect this - also could make tracing with HW2 tricky. Perhaps putting BlueRetro in debug mode should not route key presses to GamePad/GamePadAlt code.

darthcloud, 2025-01-25:
Better than first one but they both still look weird somehow.

It look like you have the blueretro connected simultaneously to the web config and the controller. And the controller data is mostly lost. Not sure I understand how it's possible.

For good measure I tried the same firmware universal hw1 v25.01 over USB only and didnt see issue.

Maybe try again making sure of the following:
*After enabling debug, close browser.
*power off BlueRetro then turn on
*Connect controller
*Press buttons
*turn controller off
*connect to web to download file

Raton-Laveur, 2025-01-25:
I noticed another unwanted behavior during my trace attempts which might explain your observations: BlueRetro and 8bitdo arcade stick unpair on 25.01 after 20 seconds, even without interaction. 

30sec video and corresponding trace attached. 

[br_debug_trace-8bitdo arcade stick-noaction-20secUnpair.zip](https://github.com/user-attachments/files/18548341/br_debug_trace-8bitdo.arcade.stick-noaction-20secUnpair.zip)

https://github.com/user-attachments/assets/05364629-4fb2-47cb-804e-eca52ca9bea8

darthcloud, 2025-01-27:
@Raton-Laveur Do you means that on v24.10 it didn't disconnect after 20sec?

darthcloud, 2025-01-27:
I feel like the mapping is what I would expect.

I understand you expect it to map to a 6 button controller.

But since fightstick can be equaly used with a 4button and 6 button system, I feel it's better for the default to be 4 buttons based as this is the more used use case.

I think it's a case where it's acceptable that manual remapping is required.

Raton-Laveur, 2025-01-27:
> [@Raton-Laveur](https://github.com/Raton-Laveur) Do you means that on v24.10 it didn't disconnect after 20sec?

Yes - I would like to confirm but HW1 25.01 disconnects during downgrade to 24.10 at 5%.

Raton-Laveur, 2025-01-27:
> I understand you expect it to map to a 6 button controller.

To Mega Drive, yes - but Saturn is 8 buttons. 
 
> But since fightstick can be equaly used with a 4button and 6 button system, I feel it's better for the default to be 4 buttons based as this is the more used use case.

Regarding the mapping: 

There is currently no arcade stick supported by BlueRetro, and arcade stick layouts changed with the 6th generation of consoles to what we have today. 

[PS1](https://www.lukiegames.com/asciiware-arcade-joystick-playstation-1-2) and [PS2](https://www.ps3blog.net/2007/03/08/ps2-arcade-stick-works-with-the-ps3/), 4x2 button layout:
L1 🔳🔺R1
L2 ❌⭕ R2

[PS3](https://archiwum.allegro.pl/oferta/wireless-arcade-stick-tekken-6-hori-pad-ps3-pc-i8160839953.html) to any Bluetooth arcade stick, 4x2 button layout:
🔳🔺R1 L1 
❌⭕ R2 L2

Sega Mega Drive / Saturn layout, observed with 8bitdo M30 or Switch Mega Drive : 
🔳🔺L1 R1 
❌⭕L2 R2

So using a Bluetooth arcade stick with pre-6th gen systems (which are the targets of BlueRetro) gives non-expected layouts.

> I think it's a case where it's acceptable that manual remapping is required.

Adding an "arcade stick layout" to these video game systems would make a whole category of arcade controllers "plug and play" in BlueRetro.

Current use case for Mega Drive / Saturn, with switching from 8bitdo M30/Switch Mega Drive controller to arcade stick depending on the game would need to manually remap keys between each game to switch layout. Having BlueRetro recognize arcade stick and map  keys as needed would be ideal.

darthcloud, 2025-01-27:
BlueRetro has a single default that is applied to every system, so making one that is plug & play for 6 button system will make it non plug & play for 4 buttons system like snes and neo geo.

So IMHO the lesser evil is to leave things as-is.

Raton-Laveur, 2025-01-27:
> Just use usb to change version since your adapter have one.

Thank you. Downgraded HW1 to 24.10 through USB, can confirm connection to 8bitdo arcade stick is stable.

darthcloud, 2025-01-27:
Do you think you connection issue are related to having debug mode enabled? Do you have connection issue with debug off on v25.01?

Raton-Laveur, 2025-01-27:
No connection issue on 25.01 debug mode off - sorry I didn't check sooner.

darthcloud, 2025-01-28:
[v25.04-beta-12-g8dc95fcc_hw1_universal.zip](https://github.com/user-attachments/files/18567909/v25.04-beta-12-g8dc95fcc_hw1_universal.zip)

This should map the way you wanted I think. Tested on on my 8Bitdo NES30 Arcade. (Which is basicly the same thing)

Make sure to use xinput mode.

On a Saturn the mapping will be:
X Y Z L
A B C R

darthcloud, 2025-01-29:
Here a more complete build sets with this:
https://github.com/darthcloud/BlueRetro/actions/runs/13023080800

closed 2025-02-17 by darthcloud