darthcloud/BlueRetro #766: Trigger R not working in gamecube game Wario World

pablopsw, 2023-09-04:
### BlueRetro firmware version

v1.8.4

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

Myself

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Original Dualshock 3 and 4 controller

### What is problem? (only list ONE problem per report)

R trigger does not work in gamecube game Wario World, I tried different settings in advanced settings, placing R trigger on other buttons and the problem persists.

### What did you expect to happen?

that the trigger R worked

### Attach files like logs or Bluetooth traces here

https://github.com/darthcloud/BlueRetro/assets/75383650/f4bd35ab-623d-41fb-8a48-67efabac34c0

[image]

kirimui, 2023-09-09:
I have the same problem, I was on 1.8.3, I updated to 1.8.4 and it still doesn't work.

rattboi, 2023-09-19:
I've tried this now, to confirm. It's weird, but yeah it doesn't work. 

I used the gc-controller-test app and it all looks right there, with merged triggers. I can't tell the difference in that app between wired gc controller and bluetooth one.

darthcloud, 2023-09-20:
So I tested this myself in v1.8.4 with PS5, PS4, PS3 and Switch controllers. I had the merged trigger preset loaded and I was able to dash with R and also Slam the ground with A then R.

Can you guys run homebrew?

If so run this https://github.com/corenting/GC-Controller-Test/releases/tag/v1.2

and take a picture while you hold R to maximum and post here. I want to see the analog value.

pablopsw, 2023-09-23:
It's the only game that happens to me. Thanks for your time.
[image]
[image]
[image]

darthcloud, 2023-09-24:
@pablopsw base on screenshot of first post you don't have the merge trigger preset that the gamecube end trigger require.

See https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#33---example-preset-load

With that preset the L & R end button will get pressed when the analog value reach >= 197

pablopsw, 2023-09-24:
It worked! I needed to read a little more, thank you for answering me and solving the problem.

darthcloud, 2023-09-24:
Closing as original author problem is fixed, but @kirimui & @rattboi feel free to post your analog value and screenshot of mapping.

closed 2023-09-24 by darthcloud

sushimehl, 2024-03-01:
Sorry but I still have this problem with my 8bitdo Pro2... I guess it's like this because i always need to change the mapping of ABYX again to work correctly, as shown on my screenshot, every time I try the merge trigger preset... Strangely enough, this is only a problem on my Gamecube, on my Wii the R button works flawless with this Game... But I still need to keep the Button mapping like on my screenshot. Any clue why?
[image]