darthcloud/BlueRetro #772: Controllers pair, but no input response

icd2k3, 2023-09-15:
### BlueRetro firmware version

1.8.4 & 1.8.3

### BlueRetro firmware specification

Select a spec

### BlueRetro firmware variant

Select a variant

### BlueRetro hardware type

Select a type

### Manufacturer

Espressif

### System used

Select a system

### Bluetooth controller brand & name

PS4, PS5, & Switch Pro

### What is problem? (only list ONE problem per report)

### TL;DR
Controllers pair fine (and I can trigger [button combination functions](https://github.com/darthcloud/BlueRetro/wiki#6---button-combinations-functions)), but inputs do not work in-game.

### Details
What's really strange is that this _**used to work**_, so I don't what I might have changed to break it... I believe it was working until I accidentally triggered the "Disconnect controller" while trying to test IGR. I also took the system apart around the same time, so it's possible I shorted something in the process too, but it wouldn't really make sense that it turns on and pairs fine.

I've tried:
- Different controllers (PS4/5/Switch Pro)
- Different PS1s (by transferring the controller module to another system)
- Checked all solder points for continuity
- Different controller module ribbon cable (to rule out potential bad one)
- Erasing and re-flashing the board via Flash Download Tool
- Multiple firmware versions (and universal vs ps1-specific)
- Advanced settings to manually set system to PS1 and correct controller mapping

Any ideas for other things I could check or try? Really at a loss after running through that list for what it might be. Only thing I can think at this point is to try a different esp32 board.

### What did you expect to happen?

That button inputs for the console would work once paired

### Attach files like logs or Bluetooth traces here

- [Video of the problem](https://www.dropbox.com/scl/fi/plp4d647ix772c1vgslvn/IMG_1117.MOV?rlkey=tt4e0ib7qr12yfk0eg0ni6fru&dl=0)
- [Install pics & video of it previously working](https://www.dropbox.com/sh/qqkhb3312wavnfh/AAANem-o0DS_4GoRl0AEvi1Za?dl=0)
- [Putty logs](https://www.dropbox.com/scl/fi/suxduemzrggfuer1m8629/putty-readout.txt?rlkey=nbj4qen0x74ml7le3o3ywrizz&dl=0)

icd2k3, 2023-09-15:
Update: Analog mode maybe? If it is, I can't figure out how to toggle it off (according to the mapping chart it should be the PS button, but pressing it doesn't fix the issue)

khtp, 2023-09-17:
Have you tried other controls? I have PS2 and in PS2 firmware 1.8.4 HW1 it works without problems for me using Xbox One and Xbox Series controls, in my case I am using everything externally, that is, outside the board and the configuration is not PSX-External but PSX-PS2. bin or something like that that comes within 1.8.4 and it works without problems where I had problems with arcade control using an esp32, my problem is published but if I use xbox one control all the buttons work and in arcade only 2 buttons fail for me L2 And R2

darthcloud, 2023-09-30:
In the config is it possible you got GamePadAlt set?

darthcloud, 2023-09-30:
Can you provided picture of the install? I wonder if one of the ESP32 IO got fried?

icd2k3, 2023-09-30:
> Have you tried other controls?
I've tried PS4, 5, and Switch Pro controllers

> In the config is it possible you got GamePadAlt set?
I'll double check in a couple days - but I don't think so

> Can you provided picture of the install? I wonder if one of the ESP32 IO got fried?
Sure! 
[image]
[image]

I know someone is working on a BR drop-in mod for the PS1 much like LaserBear's GC solution, so maybe I'll just wait for that too

icd2k3, 2023-10-05:
Here's my current config: 

[image]

darthcloud, 2025-04-22:
Is this still an issue?

closed 2025-04-22 by icd2k3