darthcloud/BlueRetro #776: 8BitDo N64 ModKit BT Trace

keanuyves, 2023-09-19:
### BlueRetro firmware version

1.8.4 Gamecube

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Retroscaler BlueRetro

### System used

Nintendo Wii Extension

### Bluetooth controller brand & name

8BitDo N64 ModKit

### What is problem? (only list ONE problem per report)

It doesn't work to remap the buttons via blueretro.io

I want to be able to use the N64 controller on the Nintendo Wii (with the dongle in the Gamecube port) to play N64 games on the Virtual Console.

### What did you expect to happen?

Being able to remap the buttons.

### Attach files like logs or Bluetooth traces here

[8BitDo ModKit - Buttons Sequence.txt](https://github.com/darthcloud/BlueRetro/files/12662152/8BitDo.ModKit.-.Buttons.Sequence.txt)
[https://drive.google.com/file/d/1N214HoMscZLQ_FAMapWLEAw9eeADIWoq/view?usp=sharing](url)

darthcloud, 2023-10-07:
Thanks for trace.

I added support for the D mode (switch mode is broken)

You can try v1.9 beta here:
https://github.com/darthcloud/BlueRetro/actions/runs/6437347093

See pairing instruction in the beta wiki:
https://github.com/darthcloud/BlueRetroWiki/blob/master/Controller-pairing-guide.md#15---8bitdo-n64-modkit-paring-guide

keanuyves, 2023-10-07:
Hello, thank you so much!

On the original N64 with the dongle, it worked without any major problems. It's a shame it doesn't have rumble support.

On the Wii, with the Gamecube dongle, if you configure only the analog trigger or only the digital trigger (L or R), it will not work in Virtual Console games. It was necessary to configure both Digital and Analog simultaneously (remap artached).

For homebrews, emulators and Wii games compatible with the Gamecube controller, works without this issue.

[image]

darthcloud, 2023-10-07:
For GC/Wii this is normal.

N64 is not the GC native controller so the default mapping can't be right. You need to make a custom config.

closed 2023-10-07 by keanuyves