darthcloud/BlueRetro #779: Atari VCS Wireless Classic Joystick - Trace

lamprey13, 2023-09-24:
### BlueRetro firmware version

N/A

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

N/A

### System used

Atari 2600

### Bluetooth controller brand & name

Atari VCS Wireless Classic Joystick

### What is problem? (only list ONE problem per report)

Doesn't pair

### What did you expect to happen?

Pair

### Attach files like logs or Bluetooth traces here

[Atari VCS Wireless Classic Joystick.zip](https://github.com/darthcloud/BlueRetro/files/12707551/Atari.VCS.Wireless.Classic.Joystick.zip)
[image]

Button Press order:

1.  A - Top Fire Button
2. B - Side Fire Button
3. Fuji Button
4. Back Button
5. Menu Button
6. Joystick Up
7. Joystick Right
8. Joystick Down
9. Joystick Right
10. Paddle/Analog Right
11. Paddle/Analog Left

This joystick also has an X-input mode. I was able to get into that mode a couple of days ago, but cannot now. I may have been connceted via USB at the time. Just wanted to mention it incase.. I'll attempt to get into that mode again and post a trace if I'm able to.

I'd offer to send you a controller, but they seem to be sold out.

RetroAsylum79, 2024-06-19:
To change mode you just press and hold the two small buttons below the joystick for ~4 seconds the fuji button will flash then the ring of lights around the joystick will flash. In VCS mode the fuji button is illuminated. In X-input the lights on the fuji button are off. In Steam using X-input mode the spinner does not work. In VCS mode the spinner works in newest version on Stella though. Hope that helps, I'd also love to see support for this as I've seen people working on blueretro based mini receivers to be used on Atari 2600 being worked on.

darthcloud, 2024-12-16:
Can you try this build?
[v25.01-beta-15-g896283a_hw1](https://github.com/darthcloud/BlueRetro/actions/runs/12352702083/artifacts/2325926437)

Should fix both Classic and Modern Atari VCS controllers.

RetroAsylum79, 2024-12-16:
Hi there, Im getting a 404 when I click on the link provided. I'll definitely give this a try. Thank you for looking into this further.

darthcloud, 2024-12-16:
You need to login into github for the link to work

RetroAsylum79, 2024-12-16:
Ah, ok sorry to bug you.

darthcloud, 2025-01-24:
Fixed in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)

closed 2025-01-24 by darthcloud