darthcloud/BlueRetro #780: PS2/PSX universal firmware - system detection issue

jerzyjurgiel, 2023-09-25:
### BlueRetro firmware version

1.8.4

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

Me :)

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Dualshock 4

### What is problem? (only list ONE problem per report)

I'm using pcb's made by pmgducati. When I plug in both cables into ps2 and connect using bluetooth webui, sometimes BlueRetro detecs PS2 as CD-I or Wii Ext. I checked PCB's and connections and everything seems solid.
I tried using 8.5V for power and 3.3V (not at the same time) but issue persists.

### What did you expect to happen?

BlueRetro detects ps2 as ps2 always.

DJm00n, 2023-10-01:
My console seems have blown vibration motor fuse - and in this case PS2 detection is failing too.
How to check - BlueRetro not working without external power from USB. Or just connect wired controller and test vibration.

jerzyjurgiel, 2023-10-03:
Vibration is working just fine on mine.
For the moment I switched from 3V to 8.5V and it seems stable.

closed 2023-10-03 by jerzyjurgiel