darthcloud/BlueRetro #781: BLUERETRO NES 2 player not functioning some games like contra

NedMapagmahal, 2023-09-25:
### BlueRetro firmware version

1.8.4

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

aliexpress

### System used

Nintendo Famicom NES

### Bluetooth controller brand & name

DUALSHOCK 4

### What is problem? (only list ONE problem per report)

blueretro works with player 1
but when playing 2 player games it having problems

for example playing contra, 2 player doesn't work
bomberman 2, 2nd player keeps pausing ang pressing bomb
nightmare on elm street the 2nd player keeps pausing
supermario 1 luigi works except he jumps to short

### What did you expect to happen?

fix the 2nd player

darthcloud, 2023-11-03:
I tested all the game you listed and player 2 work fine in all of them. No random press observed.

Are you using a NES or a Famicom?

Also I do not have any control about the quality of the stuff on AliExpress. I think your cable for player two is defective.

NedMapagmahal, 2023-11-12:
I'm using a famiclone, that has nes port, I'm not sure if it has something to do with the port, i played double dragon 2 but there's no problem with it

i tried to reset the core, with selecting none for mode, my everdrive automatically press with start button