darthcloud/BlueRetro #785: 8BitDo Retro Mechanical Keyboard not working on GameCube!

ABOhiccups, 2023-09-28:
### BlueRetro firmware version

v1.8.4

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

Internal install modkit

### Manufacturer

Laser Bear Industries

### System used

Nintendo GameCube

### Bluetooth controller brand & name

8BitDo Retro Mechanical Keyboard

### What is problem? (only list ONE problem per report)

Keyboard is not typing on Phantasy Star Online Episode I & II.

### What did you expect to happen?

Use 8BitDo Retro Mechanical Keyboard in Bluetooth Mode as alternative to Japanese GameCube Keyboard Controller.

darthcloud, 2023-12-05:
Should work in latest beta:

https://github.com/darthcloud/BlueRetro/actions/runs/7081676820

darthcloud, 2023-12-09:
Fix is now in v1.9.1

closed 2023-12-09 by darthcloud