darthcloud/BlueRetro #794: Game Cube Blue Retro Shoulder button error

SinisterSticks, 2023-10-10:
### BlueRetro firmware version

Unsure.  

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

Retro Blue

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Xbox adaptive controller and 2 Quickshot Multi-db9 selectable system in Sega Mode

### What is problem? (only list ONE problem per report)

When I map the triggers of my quick shots to the l&r of the GameCube I get a quick switch to my menu.

### What did you expect to happen?

I was expecting to drop bombs in the game Total Carnage on Midway Arcade Treasures 2.

### Attach files like logs or Bluetooth traces here

I don't know what to type here as I don't know how to access it.  The retro blue access my dongle through my Macintosh, but I typed my report on an Android.

SinisterSticks, 2023-10-19:
I think i noticed something.  I want to check if this is right...

On Total Carnage the bomb buttons are analog.  And there must be a special way I am unaware of to convert digital button presses into Game Cube L/R because there sone weird nature of the L and R triggers, where it's analog up until tge maximum, and then you can go beyond the maximum to click an internal button.

How does one get the "full click" on the Game Cube L R using "blue retro dot io"?

darthcloud, 2023-10-19:
System specific web config docs cover this:
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-System-Specific-User-Manual#mapping-15

Its even an example in the general web config:
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#33---example-preset-load

closed 2023-10-20 by darthcloud