darthcloud/BlueRetro #795: Controller Not Listed (Ipega PG-9191)

JohnBlsks, 2023-10-12:
### BlueRetro firmware version

?

### BlueRetro firmware specification

Select a spec

### BlueRetro firmware variant

Select a variant

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Retroscaler

### System used

SEGA Saturn

### Bluetooth controller brand & name

Ipega PG-9191

### What is problem? (only list ONE problem per report)

My controller is not listed

### What did you expect to happen?

Will you add support for my controller?

### Attach files like logs or Bluetooth traces here

[PG-9191 Notes.txt](https://github.com/darthcloud/BlueRetro/files/12880311/PG-9191.Notes.txt)
[image]
[PG-9191.pcapng.zip](https://github.com/darthcloud/BlueRetro/files/12880323/PG-9191.pcapng.zip)

JohnBlsks, 2023-10-13:
I forgot to press L3 & R3 the analogs down... I will redo the process tommorow and i will post again the trace... But what about the directions of the controllers i have to press them as well, or just the buttons? Thank you!

JohnBlsks, 2023-10-14:
Ok i Upload the new traces including the L3&R3, i hope i did it correct and i hope to include my controller on the Retroscaller-Blueretro wireless adapter for Sega Saturn. Thank you in advance!

darthcloud, 2025-04-16:
Feel free reopen once you got the trace.

closed 2025-04-16 by darthcloud