darthcloud/BlueRetro #796: DREAMCAST v1.8.4 beta  vmu  234P cannot use HW2

joungelee, 2023-10-12:
### BlueRetro firmware version

V1.8.4 BETA

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

Select a variant

### BlueRetro hardware type

Internal install DIY

### Manufacturer

MYSELF

### System used

SEGA Dreamcast

### Bluetooth controller brand & name

PS4

### What is problem? (only list ONE problem per report)

Use the firmware of the 1.8.4 beta version. The internal automatic switch of hw2 uses BlueRetro_ Internal_ Installation. After the firmware is painted, only 1p can be used and VMU is displayed. After the 234p is paired successfully, the handle is not displayed and cannot be controlled. It can be used after changing back to the official version 1.8.4. Is this test version incomplete or for other reasons,

### What did you expect to happen?

2 3 4pCan be used normally

darthcloud, 2023-10-13:
Post pictures of your installation where I can see all the wires.

joungelee, 2023-10-13:
> 张贴你的安装图片，我可以看到所有的电线。
Hello author, I designed my own PCB based on the schematic diagram. This PCB has been in use for a long time and should be very mature. It was used normally in V1.8.4 hw2, and 1-4p was normal. I was using the beta version v1.8.4-2-g1b74e98_ The 234P on hw2 is unusable, and the circuit has not been changed. I just downloaded the new firmware directly. I don't know if it's a firmware issue or a problem with my PCB, or because of web settings. I seem to have set both the vibration and memory card (BOTH) on the 1234P. Is this step set incorrectly?

[image]
[image]

closed 2023-10-16 by joungelee