darthcloud/BlueRetro #806: [PS2] Сharacter switches to walking

DOOTGUY, 2023-10-16:
### BlueRetro firmware version

1.9

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Original Sony Dualshock 4 (2 of them) & Switch Pro Controller

### What is problem? (only list ONE problem per report)

Сharacter switches to walking when left stick placed somewhere diagonally. Tested vid 3 controllers

### What did you expect to happen?

I expected the controller to work normally

darthcloud, 2023-10-16:
Maybe tell me the games name?

DOOTGUY, 2023-10-16:
I tested multiple games (Rayman 3, Asterix at Yhe olimpic games, Kung fu Panda) and they all have same issue, somewhere more, somewhere less, but this problem appears after update

darthcloud, 2023-10-20:
Can you try this version
[v1.9.1-beta-1-gf94313f_hw1.zip](https://github.com/darthcloud/BlueRetro/files/13048815/v1.9.1-beta-1-gf94313f_hw1.zip)

DOOTGUY, 2023-10-20:
Same problem

darthcloud, 2023-12-09:
Fix is now in v1.9.1

Make sure your adapter display version v1.9.1 in the web after updating.

I tested myself with PS4.ctrl and had not slowdown in diagonal in kungfu panda

closed 2023-12-09 by darthcloud