darthcloud/BlueRetro #817: controller (Xiaomi Mi gamepad) not in the list

Fry760, 2023-10-22:
### BlueRetro firmware version

1.9.0

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

CHINA

### System used

SEGA Dreamcast

### Bluetooth controller brand & name

Mi Gamepad MDZ-11-AA

### What is problem? (only list ONE problem per report)

Connect perfectly without problem but no input works, I was 1.8.3, 1.84 and now 1.9.0, always same issue.

### What did you expect to happen?

input to work

### Attach files like logs or Bluetooth traces here

[mi gamepad.zip](https://github.com/darthcloud/BlueRetro/files/13062613/mi.gamepad.zip)
[image]
[image]

darthcloud, 2025-01-24:
I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

Thanks

Fry760, 2025-01-27:
Nice to hear about it :) 

File in attache, I did the new update and debug-trace on my PS1 dongle, not the DC one, but got the same effect on both anyway.

[br_debug_trace.bin.zip](https://github.com/user-attachments/files/18552698/br_debug_trace.bin.zip)

darthcloud, 2025-02-01:
Can you try this beta:
https://github.com/darthcloud/BlueRetro/actions/runs/13083210355

make sure you disable debug mode before updating

Fry760, 2025-02-01:
It's working :) , just did the first level of Crash Bandicoot 3, looks like no problem. will try on the dreamcast dongle too.

closed 2025-02-17 by darthcloud