darthcloud/BlueRetro #819: [Atari Jaguar] NBA Jam inputs don't work correctly

DoomyDoomer, 2023-10-26:
### BlueRetro firmware version

1.9.4

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Humble Bazooka

### System used

Atari Jaguar

### What is problem? (only list ONE problem per report)

Starting with BlueRetro v1.7, Inputs are erratic and don't function correctly.

### What did you expect to happen?

Inputs to work as they did prior to v1.7.

darthcloud, 2023-10-27:
I think this fix it?

[v1.9.1-beta-9-gf23f16f_hw1_jaguar.zip](https://github.com/darthcloud/BlueRetro/files/13184028/v1.9.1-beta-9-gf23f16f_hw1_jaguar.zip)

darthcloud, 2023-12-09:
Fix is now in v1.9.1

closed 2023-12-09 by darthcloud