darthcloud/BlueRetro #824: Adapter losing sync/connection

mistertimely, 2023-11-02:
### BlueRetro firmware version

1.8.4

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Retrotime

### System used

Nintendo 64

### Bluetooth controller brand & name

Nintendo Switch Online N64

### What is problem? (only list ONE problem per report)

I am experiencing loss of sync/connection at seemingly random times. This includes during the firmware update process and also controller disconnects where I have to resync/reconnect my controllers. The controller stops responding and then I must press start to reconnect. Occasionally I have had to fully resync.

I am using an NTSC N64 system with a hardware v2 N64Digital installed. The updated blueretro flex already is installed and self-test shows all hearts. I am using the NSO N64 Controller.

I considered wireless interference as a possible issue and purchase controller extensions to move the adapter closer to my controller. This did not fix the issue with losing connection.

I experienced this issue also on firmware 1.8.2 prior to updating the blueretro adapter.

### What did you expect to happen?

Controller should not lose sync/disconnect on its own.

mistertimely, 2023-11-02:
To clarify, I am only using one adapter and one controller. Not multiple adapters at one time.

filthyfreak, 2023-11-04:
I am experiencing disconnection issues too. I am using the same hardware, RetroTime adapter and NSO 64 controller.

I managed to replicate the issue on command if I spam as many buttons as possible.

In other words, if I press A, B, Up/Down/Left/Right or any of the C buttons really fast, then the controller will disconnect within 5-10 seconds every time without fail.

Are you able to check to see if doing the same causes your controller to disconnect?

lbhorsman82, 2023-11-18:
I have the same issue.  Just bought and started using the retrotime 64, so I new to the device. Using NSO 64 pad also. Has anyone found a fix?

Johnson-81, 2023-11-23:
I have just sent an adapter back to Ali Express, so thought it was duff electronics as usual but the return address was a UK so thought it could have been legit. Just bougth a new one from 8BitMods in the UK and experiencing the exact same issues.

The web interface fails to connect with "Argh! Couldn't connect to BlueRetro", unable to update the firmware to 1.9 as it disconnects before completion. I have managed to get the Memory Pak working but suffer constant disconnects. Play 20 minutes of 007TWINE and was unable to finish a level as disconnects meant failing the mission. I am also getting issues where it connects to player 2 on the NSO controller.

cyatta, 2023-11-23:
Same issue after upgrade to [v1.9 hw1 n64]

What seemed to help was rolling back to the factory default firmware, which for me was  [v1.7.3 n64_external]

https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#4---system-manager-page

(Remember to dump your cpak memory cards before doing this).

I was able to marathon Next Tetris for an hour or so with zero controller drop outs.

cyatta, 2023-11-23:
Try backing up your cpak’s and doing a factory reset.

I did this on me adapaters. They’re running on 1.7.3 now. Zero connection drops after playing Tetris for an hour. On version 1.9, it happened within a few minutes.

mistertimely, 2023-11-23:
I can also confirm that after flashing 1.7.3 I haven't noticed any disconnection from the NSO controller.

Johnson-81, 2023-11-26:
I have managed to get the device connected and tried to downgrade to firmware 1.7.3, it hit 100% then hung. Power cycled and it still drops out. I'm not sure if the downgrade was successful or not. I have just spent another hour trying to get it connected and repeatedly keep getting "unable to connect" with the odd successful connection. Think it's faulty hardware to be honest as Youtibe videos I have seen connect almost instantly. Think I need to return it and stick to the Tribute pad I have :(

mistertimely, 2023-11-26:
> I have managed to get the device connected and tried to downgrade to firmware 1.7.3, it hit 100% then hung. Power cycled and it still drops out. I'm not sure if the downgrade was successful or not. [...]

Can you confirm whether you have an n64digital installed, and the pixelfx blueretro flex also installed?

Johnson-81, 2023-11-26:
Nothing installed I'm afriad. Stock PAL N64 with original power supply. I have removed my Everdrive and even the expansion pack in case they increased power consumption. Tried variations of expansion pack in/Everdrive removed, Everdrive in/Expansion pack removed etc. Definitley a genuine 8bitmods version as includes a silver sticker on the undersized of the device.

cyatta, 2023-11-26:
> I'm not sure if the downgrade was successful or not.

you should be able to see what version it’s on when you connect again via the management site.

Johnson-81, 2023-11-27:
Managed to connect to it for a split second a few minutes ago and where before it was showing "[1.8.4]", it now shows "[]". With no version number. I have just tried an original cart (V-Rally) for 10 minutes and it seemed to work fine with the NSO controller with no connection issues. I haven't tried it for any length of time, so going to have a good go today. Wonder if the Everdrive is pulling too much power causing issues.

filthyfreak, 2023-11-27:
> Managed to connect to it for a split second a few minutes ago and where before it was showing "[1.8.4]", it now shows "[]". With no version number. I have just tried an original cart (V-Rally) for 10 minutes and it seemed to work fine with the NSO controller with no connection issues. I haven't tried it for any length of time, so going to have a good go today. Wonder if the Everdrive is pulling too much power causing issues. Also before the downgrade (if successful), I have noticed some differences in the LED behaviour. The instructions mention holding for 3 seconds to pair, but it would only take holding for a second before the light would start pulsing. Likewise the factory reset would start after 3-4 seconds rather than the full 10 seconds.

Holding down the sync button on the BlueRetro N64 adapter for 10-20 seconds should reset it to the factory default firmware (1.8.3 in my case).

This fixed all my disconnection issues. Haven't had a single drop-out in my 20-40 hours play time since doing the factory reset.

What's weird is I was on this firmware version originally and was experiencing disconnection issues, so I updated to 1.9.0 in hopes that it would solve it, but 1.9.0 made it much worse. So even if you're already on 1.8.3 and are experiencing disconnects, performing the factory reset is worth a try.

This may erase your memory pak saves so back them up before doing this.

darthcloud, 2023-12-09:
Should be fix in v1.9.1

closed 2023-12-09 by darthcloud

Johnson-81, 2023-12-09:
I'll give the update a try but have never had a firmware successfully update so may not be able to update to 1.9.1. Been too busy to spend a few hours trying to update the thing. Think I'll be sending it back as the one received just doesn't seem to work, i.e. Connect long enough to send the update. Thesis for all the updates thought people

Johnson-81, 2023-12-27:
Quick update on this. I have finally managed to get the device to update to 1.9.1. Yay. I started turning the N64 off after every try, going to the start page of the web config (https://blueretro.io/), then going back into the OTA option. Just spent 15-20 minutes and managed to update the firmware. I have just connected 3-4 times on the bounce whereas before it would time out. I haven't had chance to try the NSO controller yet (starting work shortly), but fingers crossed this should stop the disconnects and losing saves. Thanks all.

captainbigbollox, 2024-10-21:
Still getting random disconnects/resets on 24.10 FW with external 8BitMods adapter.
Seems to happen most after controller pak access. Either saving or checking,