darthcloud/BlueRetro #831: Loosing Connection to GameCubes. (4 GC with 1 Controller)

LEOsMIND, 2023-11-14:
### BlueRetro firmware version

1.9

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

https://www.amazon.com/RetroScaler-Wireless-Controller-Adapter-Compatible-GameCube/dp/B0C5M8XKGY

### System used

Nintendo GameCube

### Bluetooth controller brand & name

RetroFighters N64

### What is problem? (only list ONE problem per report)

I am using the BlueRetro Adapter to control 4 GameCubes with one Controller at the same time. 
Thank you for the recent update which made that possible.

Unfortunately sometimes out of nowhere I loose the connection to one of the cubes. It's random which one, but it happens most of the time when I restart the game. I use global buttons, but also made a preset for the GameId. I need to restart the adapter to make it work again. Sometimes, also randomly, I can't control anything, but after waiting for around 10 seconds everything works again. 

It's not that big of a deal, since it does not happen that often, just wanted to let you know.

### What did you expect to happen?

A stable connection for several hours.

vaguerant, 2023-11-23:
> Sometimes, also randomly, I can't control anything, but after waiting for around 10 seconds everything works again.

If your controller identifies as a Switch controller, this second issue might be #787. There's a known issue with Switch controllers where they have random disconnects since version 1.8.4. If that's the problem affecting you, you may be able to resolve it by downgrading to 1.8.3, but presumably that will mean you can no longer control four devices simultaneously. Failing that, the disconnect issue should be resolved by the future release of version 1.9.1. That said, I suspect that none of this relates to the primary issue you describe here, only the connection drops.

darthcloud, 2023-12-09:
Can you try v1.9.1?

LEOsMIND, 2023-12-10:
Thanks for the Patch!
I played for an hour yesterday. No disconnects. I will report back after I had a longer session.

LEOsMIND, 2024-02-09:
I now tried a way more precise game and unfortunately you can see that the controller is still giving random inputs.
I tried retro fighters and PS4 controllers.

https://github.com/darthcloud/BlueRetro/assets/116081668/8b35ef80-67be-48bf-aa6f-bb2f29b1cc32

darthcloud, 2024-04-15:
ok but that sound like a different issue if you are getting random press now? (ie no disconnection?)

darthcloud, 2024-10-06:
Feel free to reopen if you got more info

closed 2024-10-06 by darthcloud